OctoPrint/OctoPrint #957: Tornado Content-Length too long uploading plugin.

foofoodog, 2015-06-29:
Uploading a plugin from the desktop fails with a "tornado.general - INFO - Malformed HTTP message from 127.0.0.1: Content-Length too long" 

1.3.0-dev-7-g5f0c89c on Pi2 from Octopi image using file Octoprint-STLViewer-master.zip (116,759 bytes) from the repo.

I have read the FAQ and I love cookies as well.

foofoodog, 2015-06-29:
Changing the non uploads maxSize in src\octoprint\settings.py fixes it.

jneilliii, 2015-06-29:
Is that on line 106?  I've had that issue before with some of the other plugins with upload capability and I think this might fix that for me as well.  Hope you like my hacked together plugin (STL Viewer).

foofoodog, 2015-06-29:
Yes, 106. I guess I will clone devel again so I can create diffs since I am back here hacking on it again.
And yes I like the STL viewer, any way to show an axis so I can get an idea of the orientation? ;)

foosel, 2015-06-29:
Should be fixed by the above commit, can someone please verify?

foofoodog, 2015-06-29:
You beat me to the patch. My way was brutal anyway. So you want I should test the a 51 MB plugin now;) Or just one more than 100k?

foosel, 2015-06-29:
more than 100k would be great ;)

foofoodog, 2015-06-29:
My git foo is so weak. I can not seem to get this code to come down.

foosel, 2015-06-29:
You are on the devel branch? And trying to pull from my repo, not maybe your fork or something like this?

foofoodog, 2015-06-29:
.git checkout devel
	Already on 'devel'
git fetch origin && git reset --hard origin/devel
	HEAD is now at 5f0c89c Fix: Ensure correct "upload date" for all added files
Doing something wrong since 5f0c89c is behind

foosel, 2015-06-29:
`git remote -v | grep origin` ?

foofoodog, 2015-06-30:
DOH!
http://gnet.homelinux.com/git/OctoPrint.git. 
So an OctoPi thing from back in April. grrr. That you provided a 1 line fix for 4 days ago (https://github.com/guysoft/OctoPi/issues/74).
I am surprised they even let me use this site given how bad I am at it.

Fix is confirmed for my scenario, thank you very much.

closed 2015-06-30 by foofoodog

jneilliii, 2015-06-30:
That fixes it for the plugin manager, but doesn't for other plugins.  Should they implement the settings change the same way within their plugin?

foosel, 2015-06-30:
@jneilliii yes, that's why there is a hook for that. having the body size limited for everything but whitelisted end points is to protect the server from malicious requests (large bodies can be especially problematic on something like a pi). OctoPrint can't possibly know about endpoints provided by plugins that should be whitelisted, hence that hook for them to tell it about them.

commit af890972 (2015-06-29): Fix: Increase max body size for plugin archives

Closes #957
commit bb0bfaee (2015-06-30): Fix: Increase max body size for plugin archives

Closes #957
(cherry picked from commit af89097)