OctoPrint/OctoPrint #960: [REQUEST] Add safety timeout feature to heating elements

fordorpi, 2015-07-06:
I would like to see code added to add a user defined timeout for the heaters when not printing.  My thought is to have a entry field to enter a time  i.e. 120 minutes for the hotend and bed heaters to be set to 0 if a print is not active.

Budman17r, 2015-07-12:
While I don't inherently disagree if you are worried about the temp after you could setup a gcode for after the print is done.  (I did that because sometimes I would set a target temp and it would stay after a print.

fordorpi, 2015-07-13:
I have this gcode set up but I was asking more for the time when you are not printing.  The other day I turned the hotend on to change filament and clean a tip and got pulled away from the printer.  I ended up leaving the hotend at 200 for two days.  Fortunately nothing was hurt other than shortening the life of my hotend but I can see how it was a fire hazard.  That is why I suggested adding logic to look to see if any printing was active and if not shut off the hotend after a defined amount of time.
I am a controls engineer and can see the PLC code for this in my head but I do not know what code is used for the Octoprint software.

foosel, 2015-07-13:
Prime example of something to be done [through a plugin](http://docs.octoprint.org/en/master/plugins/gettingstarted.html).

markwal, 2015-07-13:
The safest place to put controls like this is actually in the firmware so it also covers the case when connectivity is lost. Of course, second best would be in a plugin which wouldn't be hard to write.

fordorpi, 2015-07-13:
I would love to create a plugin if someone could help get me started in the right direction.

foosel, 2015-07-13:
Like the tutorial I linked to above?

fordorpi, 2015-07-14:
Thanks Foosel for the link and help.  I was looking at the posts on my phone and missed the link.  
How do I determine when printing is active (I actually want to know when it is not active) so I can start looking at timers to shut off the hotend?

markwal, 2015-07-14:
There's a few different ways to go about this.  One would be to reset a timer every time you saw a real gcode go by (that way, messing around with the jog controls would also reset the counter). That you would do with a gcode hook.  Or you could watch the events for start and stop print.  Look in that documentation linked to above (docs.octoprint.org) under events: http://docs.octoprint.org/en/master/plugins/mixins.html#eventhandlerplugin

aikena, 2016-02-24:
My undergrad team is going to attempt to solve this issue through a plugin if it has not already been implemented!

Thanks, 
Ashley Aiken

foosel, 2016-02-24:
@aikena perfect, marking it as grabbed then!

fordorpi, 2016-02-24:
I have not been able to put the time into this that it deserves.  Please let me know if you can get this done as I will gladly be a beta tester for this

Blue-Marlin, 2016-02-24:
http://reprap.org/wiki/G-code#M85:_Set_inactivity_shutdown_timer
At least in Marlin the default can be configured to be different from 0

aikena, 2017-10-26:
We actually did create the plugin and it worked at the time! I haven't tested it since the class, but if anyone would like to try it out or take this and work with it, feel free! https://github.com/aikena/OctoPrint-Safetytimeout

foosel, 2017-10-26:
@aikena would have been nice if you'd given us a ping here too or even better registered the plugin on the plugin repository then :)

So, turns out we now have two plugins that do this. The one linked above, and also [this one](http://plugins.octoprint.org/plugins/heatertimeout/) that was registered on the plugin repository about a month ago. So I'm closing this now.

closed 2017-10-26 by foosel

aikena, 2017-10-26:
@foosel I apologize! I think we started the process of registering the plugin but it was lost in translation along the way and I just remembered it. I am glad the issue is finally closed though! :)