OctoPrint/OctoPrint #964: [Request] add kisslicer as slice option

boelle, 2015-07-08:
Not sure if i'm right, but is cura not build in?

if so would you consider adding kisslicer also?

foosel, 2015-07-08:
I would have reopened #963 just fine if you'd given me the time...

Slicer support can be added through [plugins](http://docs.octoprint.org/en/master/plugins/mixins.html#slicerplugin). I added Cura since that is what I use myself *and* it also performs very nicely on the Raspberry Pi (which could be configured the main if not the only target platform of OctoPrint). If someone wants to add support for Kissslicer through a plugin, no problem with that, will happily add it to the repository. I won't bundle it though, not enough people using that slicer plus I don't even know if you can get it to work on a Pi. Actually, I'd rather think about un-bundling the Cura plugin in the long term instead of bundling more slicers ,)

boelle, 2015-07-08:
sorry me thought the bot would reopen once it see that i change the subject...

will have to check out if kiss at all can run on the pi..

foosel, 2015-07-08:
I had the bot do that for a while, people started manipulating it that way. So back to manual verification of validity after a first close ;)

boelle, 2015-07-08:
well they have a file for the pi.... but not the latest beta which can add mouse ears to print automatic.

will ask them when they will update and maybe even make a plugin for octoprint

closed 2019-03-19 by boelle