OctoPrint/OctoPrint #965: Can't install updates over the UI and reported available version is the same as current version

Nandox7, 2015-07-09:
Hey,

Every time I login I receive a message that there is an update available for component OctoPrint.

But when I go Settings > Software Update some issues happen:
1. The new version reported is the same as the current version (Current: 1.22 New: 1.22)
2. There is no way to perform the update, no option available
3. If I force an update it says that it was completed successfully but nothing happens (Current 1.22 / New 1.23) after it remains in 1.22.

So summarizing there are two main issues.
- It keeps reporting an update is available despite the system is running the same version
- It is not possible to perform the update using the UI

closed 2015-07-09 by GitIssueBot

foosel, 2015-07-09:
* The old version still being reported as an available update was fixed in 1.2.3
  * If I would get a dime for every time [*people do not read the instructions*](https://github.com/foosel/OctoPrint/wiki/Plugin:-Software-Update#making-octoprint-updateable-on-existing-installations) I'd be rich by now.

Nandox7, 2015-07-09:
Well I did look for documentation about it.

Went to http://octoprint.org/ > Documentation
But searching for update[s] doesn't bring anything.
(but I see now that it is on the wiki)

Still seems misleading when it returns a message saying that the update was successful, it leads to thing there is a bug/issue in the software.
It would be better to have it make the check and in case it fails to determine that he install is possible to raise it to the user and possibly link to the docs you mentioned.

foosel, 2015-07-09:
Patches welcome.