OctoPrint/OctoPrint #966: Upload button not functioning

foulowl, 2015-07-10:
1. Clicked upload button.
2. Expected dialog to open.
3. Nothing visibly happened.
4. 1.2.2.
5. Prusa i3 and Marlin release branch, but N/A in this case.
6. Debian Linux 8 and Iceweasel 31.7.0.
7. https://gist.github.com/foulowl/6ba1b5a8e1fb95611d59 (Seems N/A here)
8. serial.log is empty
9. https://gist.github.com/foulowl/92427a4bd73c9417f1eb (Seems crucial here)
10. N/A

I have read the FAQ.

Thanks!

foosel, 2015-07-10:
I don't see that javascript error, what plugins do you have installed?

foulowl, 2015-07-10:
I have:
CuraEngine
Discovery
Plugin Manager
Software Update
Virtual Printer

As an item of note, I found a work around. Clicking and dragging a file from my file browser into the Octoprint web interface uploads the file correctly, then the upload button works correctly moving forward. So perhaps Octoprint gets into a state where it can't process the "upload" button correctly due to some variable, then that variable gets reset and the upload buttons starts working again after it processes an upload. It also looks like the upload button stops working after a print completes, which might be another clue.

Thanks again!

foosel, 2015-07-10:
Is there maybe a little drop down you can expand like in this screenshot:

[image]

That would be the stack trace of the error. I need more information to pin point where that error is originating from (my guess like your's is that that's the reason for your upload button not working) but since I can't reproduce it on my end I'm pretty much flying blind.

What you could also try to maybe narrow it down is pasting this:

    [["set", "devel.webassets.bundle", false]]

into the [Yamlpatcher Plugin](http://plugins.octoprint.org/plugins/yamlpatcher/) (it's available on the repository). Should look something like this:

[image]

Then restart your server. That will cause OctoPrint to not pack up the files that make up it's frontend application anymore, so instead of packed_lib.js you'll get individual library files. That makes it slower (more things to load with the page) but it also makes it easier to pinpoint such stuff, especially with a stack trace as shown above.

You can reverse it afterwards with 

    [["set", "devel.webassets.bundle", true]]

foulowl, 2015-07-12:
Looks like I don't have that dropdown. Let me try installing YamlPatcher tonight! Thanks for the detailed debug instructions.

foulowl, 2015-07-13:
Getting: Could not import OctoPrint's setuptools, are you sure you are running that under the same python installation that OctoPrint is installed under? as both root and pi users.

foosel, 2015-07-13:
First of all, keep your hands off the root user, you don't need it for anything regarding OctoPrint (and if you do, you broke some permissions somewhere and that should be fixed instead of forcing stuff with sudo/root).

Secondly, as the message hints at, you need to install plugins with the same python installation that you installed OctoPrint under. That is probably a virtual environment, NOT the default python installation.

For OctoPi: `~/oprint/bin/pip`. If you installed manually following the guide on the wiki: `~/OctoPrint/venv/bin/pip`. Also, you can [install the plugin via the plugin manager](http://plugins.octoprint.org/help/installation/) (unless that's broken through your Javascript error too), no need to use the command line.

foulowl, 2015-07-15:
Ah, I'm used to installing packages via a system package manager (thus root) I have no experience with pip or similar yet. :)

Alrighty, let me instal YamlPatcher via the plugin interface. Thank you!

foulowl, 2015-07-15:
Alright, YamlPatcher installed, [["set", "devel.webassets.bundle", false]] dumped into YamlPatcher, Octoprint restarted, yet it looks like I still see the same output in iceweasel's "browser console":
[image]

When I click on "packed_libs.js" I get the following line:
`(function(e,undefined){var t,n,r=typeof undefined,i=e.location,o=e.document,s=o.documentElement,a=e.jQuery,u=e.$,l={},c=[],f="2.0.0",p=c.concat,h=c.push,d=c.slice,g=c.indexOf,m=l.toString,y=l.hasOwnProperty,v=f.trim,x=function(e,n){return new x.fn.init(e,n,t)},b=/[+-]?(?:\d*\.|)\d+(?:[eE][+-]?\d+|)/.source,w=/\S+/g,T=/^(?:(<[\w\W]+>)[^>]*|#([\w-]*))$/,C=/^<(\w+)\s*\/?>(?:<\/\1>|)$/,k=/^-ms-/,N=/-([\da-z])/gi,E=function(e,t){return t.toUpperCase()},S=function(){o.removeEventListener("DOMContentLoaded",S,!1),e.removeEventListener("load",S,!1),x.ready()};x.fn=x.prototype={jquery:f,constructor:x,init:function(e,t,n){var r,i;if(!e)return this;if("string"==typeof e){if(r="<"===e.charAt(0)&&">"===e.charAt(e.length-1)&&e.length>=3?[null,e,null]:T.exec(e),!r||!r[1]&&t)return!t||t.jquery?(t||n).find(e):this.constructor(t).find(e);if(r[1]){if(t=t instanceof x?t[0]:t,x.merge(this,x.parseHTML(r[1],t&&t.nodeType?t.ownerDocument||t:o,!0)),C.test(r[1])&&x.isPlainObject(t))for(r in t)x.isFunction(this[r])?this[r](t[r]):this.attr(r,t[r]);return this}return i=o.getElementById(r[2]),i&&i.parentNode&&(this.length=1,this[0]=i),this.context=o,this.selector=e,this}return e.nodeType?(this.context=this[0]=e,this.length=1,this):x.isFunction(e)?n.ready(e):(e.selector!==undefined&&(this.selector=e.selector,this.context=e.context),x.makeArray(e,this))},selector:"",length:0,toArray:function(){return d.call(this)},get:function(e){return null==e?this.toArray():0>e?this[this.length+e]:this[e]},pushStack:function(e){var t=x.merge(this.constructor(),e);return t.prevObject=this,t.context=this.context,t},each:function(e,t){return x.each(this,e,t)},ready:function(e){return x.ready.promise().done(e),this},slice:function(){return this.pushStack(d.apply(this,arguments))},first:function(){return this.eq(0)},last:function(){return this.eq(-1)},eq:function(e){var t=this.length,n=+e+(0>e?t:0);return this.pushStack(n>=0&&t>n?[this[n]]:[])},map:function(e){return this.pushStack(x.map(this,function(t,n){return e.call(t,n,t)}))},end:function(){return this.prevObject||this.constructor(null)},push:h,sort:[].sort,splice:[].splice},x.fn.init.prototype=x.fn,x.extend=x.fn.extend=function(){var e,t,n,r,i,o,s=arguments[0]||{},a=1,u=arguments.length,l=!1;for("boolean"==typeof s&&(l=s,s=arguments[1]||{},a=2),"object"==typeof s||x.isFunction(s)||(s={}),u===a&&(s=this,--a);u>a;a++)if(null!=(e=arguments[a]))for(t in e)n=s[t],r=e[t],s!==r&&(l&&r&&(x.isPlainObject(r)||(i=x.isArray(r)))?(i?(i=!1,o=n&&x.isArray(n)?n:[]):o=n&&x.isPlainObject(n)?n:{},s[t]=x.extend(l,o,r)):r!==undefined&&(s[t]=r));return s},x.extend({expando:"jQuery"+(f+Math.random()).replace(/\D/g,""),noConflict:function(t){return e.$===x&&(e.$=u),t&&e.jQuery===x&&(e.jQuery=a),x},isReady:!1,readyWait:1,holdReady:function(e){e?x.readyWait++:x.ready(!0)},ready:function(e){(e===!0?--x.readyWait:x.isReady)||(x.isReady=!0,e!==!0&&--x.readyWait>0||(n.resolveWith(o,[x]),x.fn.trigger&&x(o).trigger("ready").off("ready")))},isFunction:function(e){return"function"===x.type(e)},isArray:Array.isArray,isWindow:function(e){return null!=e&&e===e.window},isNumeric:function(e){return!isNaN(parseFloat(e))&&isFinite(e)},type:function(e){return null==e?e+"":"object"==typeof e||"function"==typeof e?l[m.call(e)]||"object":typeof e},isPlainObject:function(e){if("object"!==x.type(e)||e.nodeType||x.isWindow(e))return!1;try{if(e.constructor&&!y.call(e.constructor.prototype,"isPrototypeOf"))return!1}catch(t){return!1}return!0},isEmptyObject:function(e){var t;for(t in e)return!1;return!0},error:function(e){throw Error(e)},parseHTML:function(e,t,n){if(!e||"string"!=typeof e)return null;"boolean"==typeof t&&(n=t,t=!1),t=t||o;var r=C.exec(e),i=!n&&[];return r?[t.createElement(r[1])]:(r=x.buildFragment([e],t,i),i&&x(i).remove(),x.merge([],r.childNodes))},parseJSON:JSON.parse,parseXML:function(e){var t,n;if(!e||"string"!=typeof e)return null;try{n=new DOMParser,t=n.parseFromString(e,"text/xml")}catch(r){t=undefined}return(!t||t.getElementsByTagName("parsererror").length)&&x.error("Invalid XML: "+e),t},noop:function(){},globalEval:function(e){var t,n=eval;e=x.trim(e),e&&(1===e.indexOf("use strict")?(t=o.createElement("script"),t.text=e,o.head.appendChild(t).parentNode.removeChild(t)):n(e))},camelCase:function(e){return e.replace(k,"ms-").replace(N,E)},nodeName:function(e,t){return e.nodeName&&e.nodeName.toLowerCase()===t.toLowerCase()},each:function(e,t,n){var r,i=0,o=e.length,s=j(e);if(n){if(s){for(;o>i;i++)if(r=t.apply(e[i],n),r===!1)break}else for(i in e)if(r=t.apply(e[i],n),r===!1)break}else if(s){for(;o>i;i++)if(r=t.call(e[i],i,e[i]),r===!1)break}else for(i in e)if(r=t.call(e[i],i,e[i]),r===!1)break;return e},trim:function(e){return null==e?"":v.call(e)},makeArray:function(e,t){var n=t||[];return null!=e&&(j(Object(e))?x.merge(n,"string"==typeof e?[e]:e):h.call(n,e)),n},inArray:function(e,t,n){return null==t?-1:g.call(t,e,n)},merge:function(e,t){var n=t.length,r=e.length,i=0;if("number"==typeof n)for(;n>i;i++)e[r++]=t[i];else while(t[i]!==undefined)e[r++]=t[i++];return e.length=r,e},grep:function(e,t,n){var r,i=[],o=0,s=e.length;for(n=!!n;s>o;o++)r=!!t(e[o],o),n!==r&&i.push(e[o]);return i},map:function(e,t,n){var r,i=0,o=e.length,s=j(e),a=[];if(s)for(;o>i;i++)r=t [... cut ...] equeue(e,t)};"inprogress"===i&&(i=n.shift(),r--),o.cur=i,i&&("fx"===t&&n.unshift("inprogress"),delete o.stop,i.call(e,s,o)),!r&&o&&o.empty.fire()},_queueHooks:function(e,t){var n=t+"queueHooks";return q.get(e,n)||q.access(e,n,{empty:x.Callbacks("once memory").add(function(){q.remove(e,[t+"queue",n])})})}}),x.fn.extend({queue:function(e,t){var n=2;return"string"!=typeof e&&(t=e,e="fx",n--),n>arguments.length?x.queue(this[0],e):t===undefined?this:this.each(function(){var n=x.queue(this,e,t);`

o_O

markwal, 2015-07-15:
That yamlpatcher line should have made it so you didn't have packed libraries any more. I think maybe the setting got stomped after you set it (don't click save on settings only click apply then cancel) our perhaps you didn't restart the server?

Any way, you can check by doing the patcher again. If it is still set it should say no change in the preview window. You'll know it worked when your browser error is * not* in packed_lib

foulowl, 2015-07-15:
It says "no change" in the preview window, so I know it's set.

foosel, 2015-07-15:
You have cleared your browser's cache and reloaded the page I guess? Because then something is really odd here, with that exact patch applies it definitely doesn't bundle the files on my end anymore.

It would also be great if you could try with another browser which DOES offer a stack trace, like Chromium.

Also, please don't dump the full packed library anymore :P

foulowl, 2015-07-15:
Ah, I didn't know I needed to clear my cache. I have never debugged a javascript error before. 

My browser console now reports jquery.min.js:3 as the source of the error.

Apologies for including that large amount of text, I just copy and pasted the line without looking at how long it was, otherwise I would have used a gist.

foosel, 2015-07-15:
Ok, then at least the bundling now is not active anymore, but of course, without a proper stack trace that still has us flying somewhat blind... Is there a way for you to reproduce this bug in Chromium? Because that definitely has a stack trace included with the error.

foulowl, 2015-07-15:
As of yet I have been unable to reproduce in chromium. Let me see if I can get a stack trace with firebug. (Which I have never used but it looks like it should work)

foulowl, 2015-07-16:
No luck yet. I was able to reproduce with firebug running, but looked like there was no obvious output.

Cannot reproduce in chromium, so perhaps the fix here is "don't use firefox/iceweasel."

Considering the limited time and resources of the devs, and the fact that I appear to be the only user with this issue, perhaps we should close and move on? What is your opinion?

markwal, 2015-07-16:
Apparently firebug requires a console command to dump out the stack trace.  You type this at the console: "console.trace()" and then press enter.

foulowl, 2015-07-21:
I can reproduce the bug more reliably now (it always occurs after a print has finished) but still can't get a stack trace.

Firebug shows "Empty string passed to getElementById()." on jquery.min.js:3, but running console.trace() does not provide any additional information.

However, "Empty string passed to getElementById()." appears to be a warning, not an error, so perhaps that's why we can't get a stack trace?

markwal, 2015-07-21:
I imagine firebug doesn't stop on the warning so that's why console.trace() isn't helpful.  There is a button on the upper lefthand corner of the console pane that is "Stop on All Errors" which appears to be unset by default.  Perhaps if you activate that before reproducing?

foosel, 2015-07-21:
Ok, so I could finally reproduce that JS warning myself after some exchange/debugging session with @markwal and @Mikk36 (thanks for that!) and fixed the reason for it on the `devel` and `maintenance` branch. You might want to try either of those to see if they fix your login problem. I have to say though, I seriously doubt it, since it sounds more like something else is going on. I have no idea what though, so let's try it anyway.

foulowl, 2015-07-24:
I only have one system running octoprint right now. Let me get a second system spun up and I'll test the devel and maintenance branches. Thank a bunch! 

BTW, If this bug isn't affecting anyone else, I have no problem with simply moving on, I can use curl for uploads as a workaround. I understand dev time is limited, and I would rather not bog down the issues with something minor if it only affects me. 

Thanks again!

foosel, 2015-07-24:
The fix has also made it to the `master` branch in the meantime, I released it as part of 1.2.4 yesterday. If you happen to have updated already, it should be active.

Also, a big thank you for understanding that us devs sometimes can feel like between a rock and a hard place with issues like these. I'd still like to get to the bottom of this, because if you are experiencing it, others might be affected to and just not bother to come here, but it's nice to be talking to someone who understands the general picture too ;)

foulowl, 2015-07-24:
Awesome! Let me update to 1.2.4 after my current print is finished and I'll see if I can reproduce.

Of course! I'm guessing that you and the other Octoprint devs are volunteers, like many other open source projects, so a huge thank you to all of you! I hope I can help contribute by tracking down and reproducing bugs that I find as the end-user.

foosel, 2015-07-24:
In my specific case I can say that thankfully my work is funded by spanish technology company BQ since last August. I can work full time on this without having to worry about how to pay my rent and food. Regardless of this it is nevertheless always tough to make the right call when it comes to trying to track down errors encountered by individuals in the field vs. working on things profiting the other 99% of the user base, and not many people understand this. So - thanks for not being one of them ;)

foulowl, 2015-07-31:
That is awesome, glad to hear there is some money behind Octoprint, it is a fantastic tool. 

Unfortunately it looks like the issue persists in 1.2.4. However, I am fine with using the API to upload files, so feel free to close or I can continue to troubleshoot on my end. :)

I have noticed as well that this issue will almost always occur after a print has completed. (Probably 75% of the time)

foosel, 2017-03-08:
@foulowl do you still see this under 1.3.1? I haven't been able to reproduce this and nobody else has reported similar problems.

foosel, 2019-11-05:
Considering this closed... still unreproduced and no reports to the contrary over the past two years.

closed 2019-11-05 by foosel