OctoPrint/OctoPrint #967: Octoprint does not remember status of certain checkboxes

foulowl, 2015-07-10:
1. Loaded Octoprint web interface. This issue occurs after an Octoprint restart, a browser restart, or sometimes if web interface idles for awhile. Checkboxes affected: all "Renderer options" and the two "suppress" checkboxes.
2. Expected Octoprint to remember the checkbox status on a per user basis. (Also I only have one user on my system)
3. Nothing visibly happened.
4. 1.2.2.
5. Prusa i3 and Marlin release branch, but N/A in this case.
6. Debian Linux 8 and Iceweasel 31.7.0, also occurs on an Android phone and on a Debian 7 system.
7. https://gist.github.com/foulowl/6ba1b5a8e1fb95611d59 (Seems N/A here)
8. serial.log is empty
9. N/A
10. N/A

I have read the FAQ. 

Thank you devs for all your hard work. I hope I am helping by catching bugs.

markwal, 2015-07-10:
I think this is a duplicate of the following and the related issues: https://github.com/foosel/OctoPrint/issues/443

foosel, 2015-07-10:
Jup

closed 2015-07-10 by foosel