OctoPrint/OctoPrint #968: Temp graph not updating (1.2.3 master branch)

smartroad, 2015-07-12:
Hi! I have recently update my OctoPi to the latest 1.2.3 version. I am having one issue with the temperature graph and am not sure if it is Octoprint or Repetier Firmware (0.92).

When I set the temperature it updates to reflect that but then doesn't continue. Checking the terminal i get:
```
Send: M140 S60.000000
Recv: ok 0
Recv: TargetBed:60
Recv: wait
Recv: wait
Send: M105
Recv: ok 0
Recv: T:23.05 /0 B:23.33 /60 B@:255 @:0 T0:23.05 /0 @0:0 T1:22.80 /0 @1:0
Recv: wait
Recv: wait
Recv: wait
Recv: wait
Recv: wait
...
```

And it just sits there with Recv: wait. Eventually there is sometimes and update but otherwise it is just the above.

foosel, 2015-07-12:
Duplicate of #835 which also includes the solution. You need to disable external heat up detection under Features (one of those options with a "Repetier" tag) since Repetier is a special snowflake.

foosel, 2015-07-12:
Relevant comment: https://github.com/foosel/OctoPrint/issues/835#issuecomment-95835678

closed 2015-07-12 by foosel

smartroad, 2015-07-12:
Cheers! Was just about to update this that I had found the very same post :grin: (though ended up going through Google and not the sites search :cry: )

Again thanks and sorry for the time waste :+1: