OctoPrint/OctoPrint #969: [Request] Support compressed file uploads

ermahgerdgertherb, 2015-07-14:
It would be really nice if Octoprint accepted compressed file uploads.  The difference is noticeable, particularly when the server's network performance is not that great:

 7397526 Jul 13 19:55 BS_Joint_9x.gcode
 1328432 Jul 13 20:09 BS_Joint_9x.gcode.xz

Seemingly if there's a "load and print" then that's the point where it can decompress it...

foosel, 2015-07-28:
The loading step is the wrong point for decompression. It should rather be part of an upload post processing step.

While I agree that it would make sense to add this to the core application, it could btw also already be added through a plugin.

ermahgerdgertherb, 2015-07-30:
Alrighty.

So I guess I can add .xz as an a machine file extension via octoprint.filemanager.extension_tree, and then unpack it with octoprint.filemanager.preprocessor (I don't see a literal "post processing" hook), return a new file object with the .xz removed, pointing at the uncompressed file.

Hypothetically, would this plugin play together with a strip_comments plugin?  Like if I uploaded "foo_strip.gcode.xz", would it decompress that and then run the strip_comments on it?  Or how about a completely fictitious "foo.gcode.xz.xz.xz"?  Is it turtles all the way down?

Also, the docs for the preprocessor hook look like they're cut off at "Handlers which do not wish to handle the file_object"...