OctoPrint/OctoPrint #971: Timelapse downloads get delayed

Mikk36, 2015-07-15:
There seems to be something wrong when Octoprint is downloading the images for timelapse from mjpg-streamer, because it often gets delayed by many seconds.
Here's a sample log file showing that:
https://dl.dropboxusercontent.com/u/54281287/octoprint%20%282%29.log
Out of 3 timelapses, first is with settings 60 fps, 2 second post-roll, 1 second interval and the other two with 4 second interval.
As a comparative test, I made a little script for node.js to download 120 images exactly at 1 second intervals, which got me 120 images with some 2 millisecond variance.
Here's the script file and also sample images recorded with the script: https://dl.dropboxusercontent.com/u/54281287/jpgStreamerTest.rar
This showed me that mjpg-streamer is not to blame, because it can deliver images on time.

foosel, 2015-07-15:
Next time please copy the bug reporting template

foosel, 2015-07-20:
Ok, so... I refactored the whole thing so that the captures are now done sequentially from within a queue, and so that it's guaranteed no captures are still pending when the movie begins to render - that should solve at least the problem of frames from the post roll getting lost due to any delays and also reduce the number of concurrent active threads, hence hopefully mitigating the delay issue a bit. See commit 4f5dc708283ac9703c617de81cea3bc93e3abf1e for that.

What I can't do anything about is that it's not guaranteed at all that the thread responsible of performing the captures will get to it right away, so depending on how busy other parts of the application (or the system) are, stuff might get delayed - this ain't a real time system, scheduling is happening in places way beyond my control. There's nothing at all I can do about that on single-core systems like the Pi1, and even on the Pi2 I'd need to completely switch over to the `multiprocessing` module (to get around Python's GIL) in order to *maybe* make it less prominent. 

Frankly, I'm not seeing how a bunch of delayed timelapse frames would justify the amount of work that would need to be put into that and keep me from doing other things that are of more benefit to the users, so I'm closing this.

closed 2015-07-20 by foosel