OctoPrint/OctoPrint #981: RuntimeError: No pip available, can't operate

hradec, 2015-07-20:
1. Clicked install  button of a plugin.
    2. Expected dialog to open.
    3. dialog opened saying the plugin was installed suscessfully, but a pink dialog showed up at to top right corner saying" something went wrong, please consult octoprint.log for details. Pip is installed, by the way, and I can install the plugin from url using pip in the command line as user "pi" without sudo with no problems!
    5. 1.3.0 (master branch), but was also not working on 1.2.3!
    6. Custom reprap 3d printer running Marlin release branch, but N/A in this case.
    7. Centos 7  and SeaMonjey 2.33.1. (although got the same issue with chrome)
    8. http://pastebin.com/WrVDgend

I have read the FAQ.

Thanks!

markwal, 2015-07-21:
The OctoPrint plugin manager doesn't look for pip on the path, it looks for pip in the same directory as the python that OctoPrint is running under.  This is so that it works correctly in a virtualenv.  From looking at the log and the sources, my guess is that if _find_pip had found a pip in the same folder as python and it was accessible then we would have seen a warning or an error earlier in the log prior to the no pip error at the end of the log  If _find_pip succeeds it will log Found pip at {}, version is {} which I didn't see in your log.

Current recommendation for setting up OctoPrint is in a virtualenv that belongs to the pi user and that's the condition that gets the most testing.  Since your install is done as root with user pi added to site-libraries, my guess is that has something to do with either not finding pip or not having access to it in the daemon.

foosel, 2015-07-21:
Please follow the recommended install scenario, then everything will work. Easy steps (assuming Raspberry Pi and default checkout location of `~/OctoPrint`, adjust as necessary for your specific setup):

    cd ~/OctoPrint
    sudo apt-get install python-virtualenv
    virtualenv venv
    ./venv/bin/python setup.py install

Then adjust your startup script to point at `/home/pi/OctoPrint/venv/bin/octoprint` instead of (probably) `/usr/bin/octoprint` or `/usr/local/bin/octoprint` and you are done. Configuration will still be there.

Alternatively:

Since you said you made the site-packages directory writable for all users (ewww...) it should also work here, but something's odd with your setup. My guess would be you are running OctoPrint with `/usr/bin/python`, which is the default location here, and when pip is properly installed it will be located under `/usr/bin/pip`, which OctoPrint would be able to detect. It doesn't here, so something is broken. You can tell OctoPrint where pip is located, e.g.:

``` yaml
plugins:
  pluginmanager:
    pip: /my/odd/locationed/pip
```

(``[["set", "plugins.pluginmanager.pip", "/my/odd/locationed/pip"]]`` in YamlPatcher)

But I'd really suggest going the virtual env route.

closed 2015-07-21 by foosel

gaetancollaud, 2017-02-11:
Hi guys,

Sorry to dig up this issue, but I have the same problem using docker. You can see https://github.com/OctoPrint/docker/issues/2

I have followed the setup documentation. And it seems like my venv is correct.

``` sh
octoprint@fa83245f72be:/opt/octoprint/venv/bin$ ls
activate       activate_this.py  futurize     netaddr     pip     pkginfo        pyrsa-decrypt-bigfile  pyrsa-keygen    pyrsa-verify   python2    watchmedo
activate.csh   easy_install      markdown_py  octoprint   pip2    pybabel        pyrsa-encrypt          pyrsa-priv2pub  python         python2.7  webassets
activate.fish  easy_install-2.7  miniterm.py  pasteurize  pip2.7  pyrsa-decrypt  pyrsa-encrypt-bigfile  pyrsa-sign      python-config  unidecode  wheel
```

For information I start octoprint using `/opt/octoprint/venv/bin/octoprint serve`