OctoPrint/OctoPrint #983: [Request] How to make an Octoprint plugin for DLP printers.

hstarmans, 2015-07-22:
I am trying to write an Octoprint plugin for DLP printers. At the moment, this plugin is not working, you can see an early version here; https://github.com/hstarmans/Octoprint-DLPPLUGIN . 
As it is Octoprint's goal to provide a neat interface for any 3D printer. The question basically boils down to; how do I do write a plugin for any printer!? Think about food printers, large area stereolithography machines etc.
To clarify my question, I have formulated a set of sub questions:
1. How do I connect my plugin to the print, pause and cancel buttons of Octoprint!?
2, How do I disable the temperature control and GCode viewer tab? Rationale; the 3D printers I use, think large scale stereolithography machines or food printers, do not only use G-Code.
3. What should I use as Statemachine for my plugin!? Ideally, I would like to subclass something like Qstatemachine, see http://doc.qt.io/qt-4.8/qstatemachine.html. QStatemachine can be used in Python. Typically Qt is not used for websites, so what should I then  use?!
I would then like add the states for my specific printers to this machine and define the transitions. For example, my machine would have a start up state, a loop state, a final state, an abort state and a pause state. To illustrate,  two  of these states look as follows. The loop state would have the following substates; move up platform, project image on beamer, (repeat)
If I click the pause button, the machine immediately transitions to the pause state. A move would be allowed to be executed but a beamer projection interrupted. If I click pause again, the print resumes, the machine checks if it still is in the correct position and the beamer completes its projection.

The whole thing about statemachine might sound complicated but creating this is really in a statemachine. If Octoprint would have a general state machine it could allow the FDM printer be attached to it as a special case.
Different printers would simplify define different implementations of this Statemachine.

Look forward to your response! Really like Octoprint and learned a lot from the tutorials.

Best,
Rik

hstarmans, 2015-07-25:
I have looked somewhat further into it. Could I combine the Flask event loop with the Qt event loop, so I have both a Flask (for the webserver) and Qt application (for Qt stuff like QSocketnotifier, QTimer etc)!? Or would that be a bad idea? Example of a combination of these two is here; https://gist.github.com/andrewbunday/1358794

foosel, 2015-07-25:
You don't attach your plugin to any buttons, OctoPrint calls certain interfaces your plugin implements to do what the button press requires it to do. OctoPrint tracks its own state internally. You are looking at the [hollywood principle](https://en.wikipedia.org/wiki/Hollywood_principle) here.

I really don't want to curb your enthusiasm, but you are approaching this the wrong way I fear.

You have way more pressing issues on hand with that project than trying to get OctoPrint and QT to coexist next to each other. Before you can even think of tackling this, you'll need to understand how OctoPrint works - if you want to write a plugin for OctoPrint, you'll have to understand what you are coding against, not try to force it into a different corset. Look at the documentation, look at the code, understand how things work together and what options plugins have to change existing behaviour. From the questions you are asking, I'm assuming that you haven't tackled that phase yet.

You also need to figure out whether what you are trying to achieve - controlling a DLP printer through OctoPrint - is even possible with what is already there. Although people keep surprising me with what they make possible through plugins, I fear OctoPrint is not at the point yet to allow for such a radical different way of printing.

That being said, I'd also strongly advise against pulling a very very large (and also legally complicated) dependency such as PyQT/PySide into a plugin just to use one or two helper classes from it. You should try to work with what you have and make minimally invasive adjustments, otherwise you'll have a behemoth of work before you, make installation needlessly complicated for your users, etc etc. In a nutshell: my personal opinion is that you are definitely rushing into the wrong direction when you are focusing so much on what QT offers you that might be useful in terms of 3D printer communication when squinting at it from 10 feet distance.

Finally:

> It is Octoprint's goal to provide a neat interface for any 3D printer, a DLP plugin is therefore a must.

No, definitely not "any". OctoPrint's goal first and foremost is to provide its current functionality for RepRap style, FDM based 3D printers that adhere to the established GCODE based communication, and do this reliably. Yes, in the long term I also want it to be even more extendable so that something like a DLP plugin would in fact be easy to do. But the above statement would be reaching for the stars at the moment and never something I'd claim this project is about.

hstarmans, 2015-07-26:
Thank you for your reply. I agree, adding the Qt platform adds a very large dependency to Octoprint and is an invasive adjustment. On the downside, I still don't know the answer to any of my questions. 
To me, question 3 and 4 seem most complicated. 
Flask recommends for asynchronous tasks a pattern which uses [Celery] (http://flask.pocoo.org/docs/0.10/patterns/). There is an elusive example of it's use with Flask here by [Miguel Grinberg] (http://blog.miguelgrinberg.com/post/using-celery-with-flask).
Is it possible that Octoprint comes packaged with Celery!?  I can imagine, Celery can provide the functionality needed to solve question 3 and 4.
PS.
Unrelated to the topic, I advocate the use of FFF over of FDM. FDM is a trademark registered by Stratasys. A legally unconstrained term is for example [FFF] (http://lawitm.com/3d-printing-law-trademarks-why-fdm-isnt-for-everybody/)

foosel, 2015-07-26:
Your first question has been answered - you don't attach anything yourself, your plugin code gets called by OctoPrint. Your third question has been answered - OctoPrint tracks its internal state itself. If you need an additional state within your plugin, choose wisely and don't pull in large dependencies for something like a statemachine which can easily be implemented in a couple of lines yourself if need be. Your second and fourth question are the "wrong" questions - I told you, you need to understand how the platform you are targeting works first and what it allows you to do. Once you do that, you'll realize that for question 2 there exist two settings, however they are not "toggleable" by plugins, and that the whole line by line communication is already taken care of by the communication layer within OctoPrint, so the question how to "handle Carriage Returns" is a non issue.

I can only repeat what I already told you. If you are serious about giving this a try, look at how OctoPrint works, how it is bootstrapped, how the communication with the printer is achieved (hint: OctoPrint already does a lot of things asynchronously, and it doesn't need Celery for that). Go through the available extension points in the documentation, read the source. You will have to do this yourself if you are serious about tackling this, and you'll have to keep in mind that this might also just be the wrong time to achieve your goal - as I already told you, the communication layer within OctoPrint is currently focused on the necessities of FFF (better?) and not prepared to handle radically different methods of printing. it might still be possible to get it working with a DLP, but you'll need to get creative for that at the current state of the available extension points, and that means you need to (and need to want to) understand the platform you are targeting (I cannot repeat that enough).

hstarmans, 2015-07-26:
Thank you. I still can't say I understand all the answers but at least now I realize what they are.
I have looked more closely at the source code of Octoprint like [standard.py](https://github.com/foosel/OctoPrint/blob/master/src/octoprint/printer/standard.py) and [events.py](https://github.com/foosel/OctoPrint/blob/master/src/octoprint/events.py). I will focus my attention on different things for now, maybe I will come back to this problem later or just write my own dirty Qt Flask fix :).

foosel, 2015-07-26:
Closing this them for now

closed 2015-07-26 by foosel

hstarmans, 2016-02-02:
@foosel, thank you for your feedback. I have created a traffic light using Flask, Tornado and which is modelled with a statemachine, for which I used [transitions](https://github.com/tyarkoni/transitions). The traffic light runs in the background. To do so the machine is dispatched to a separate coroutine which keeps calling _next_ . For a code example, see my second example using coroutines in [issue 63](https://github.com/tyarkoni/transitions/issues/63)