OctoPrint/OctoPrint #985: API - File upload not working

moony871, 2015-07-24:
Hello,

just switched from version 1.1.2 to 1.2.4 and an exception occur if I try to upload a file through the API.

 ```
2015-07-24 23:00:24,409 - tornado.application - ERROR - Uncaught exception
Traceback (most recent call last):
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/tornado-4.0.1-py2.7-linux-armv6l.egg/tornado/http1connection.py", line 542, in _read_fixed_body
    yield gen.maybe_future(delegate.data_received(body))
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/tornado-4.0.1-py2.7-linux-armv6l.egg/tornado/httpserver.py", line 274, in data_received
    return self.delegate.data_received(chunk)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/tornado-4.0.1-py2.7-linux-armv6l.egg/tornado/web.py", line 1874, in data_received
    return self.handler.data_received(data)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/OctoPrint-1.2.4-py2.7.egg/octoprint/server/util/tornado.py", line 209, in data_received
    self._process_multipart_data(data)
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/OctoPrint-1.2.4-py2.7.egg/octoprint/server/util/tornado.py", line 254, in _process_multipart_data
    self._on_request_body_finish()
  File "/home/pi/OctoPrint/venv/local/lib/python2.7/site-packages/OctoPrint-1.2.4-py2.7.egg/octoprint/server/util/tornado.py", line 369, in _on_request_body_finish
    self._new_body += p + b"\r\n"
TypeError: unsupported operand type(s) for +: 'NoneType' and 'str'
2015-07-24 23:00:44,116 - octoprint.server.util.sockjs - INFO - Client connection closed: 192.168.178.22
```

closed 2015-07-24 by GitIssueBot

foosel, 2015-07-25:
Something about your upload request clearly is wrong. I suggest providing
this alongside the other required information to get help. Please
understand that people can't solve issues when they don't have the full set
of information.

foosel, 2015-08-11:
Since you didn't provide the request you were using, the only thing I could do was make sure the code doesn't hiccup on unset fields in the multipart headers. I hope that solves it. Will be included in 1.2.5.

commit 4bb5be76 (2015-08-10): Do not hiccup on unset Content-Type fields in multipart file uploads

Just don't set the content_type parameter in that case.

Fixes #985
commit e3286951 (2015-08-11): Do not hiccup on unset Content-Type fields in multipart file uploads

Just don't set the content_type parameter in that case.

Fixes #985
(cherry picked from commit 4bb5be7)