OctoPrint/OctoPrint #990: Add soft z-axis zeroing

sherifeid, 2015-07-26:
I would like to work on and see how feasible it is to add the ability to add an offset to the Z-axis zero. Typically Z-axis is homed using the stop switch, now over time there are little shifts due to many reasons, is it possible to add a soft-zeroing button to the control menu where the user can adjust the z-axis at 0.1u clicks then zero it

sherifeid, 2015-07-26:
I'm sorry, I'm new to github, so I don't know how to label this as a new feature

closed 2015-07-26 by GitIssueBot

Salandora, 2015-07-29:
You need to put a [Request] in front of your title but don't worry foosel can reopen it.

Oh and what you like to do can be done by custom Controls already. Atleast the soft zeroing can be done by a G92 Z0 command. 
And Z can be moved in 0.1 steps. Is this to much?

For defining custom Commands you can use my plugin or do it the config.yaml way therefore you should look at the docs: docs.octoprint.org
 
I hope I was able to help you :-)

foosel, 2015-07-29:
Not going to reopen it since as correctly pointed out this can already be done with what is available, and is probably something that should be done in firmware anyhow (M212 comes to mind)