OctoPrint/OctoPrint #992: [Request] Fast extrusion/retraction for bowden setups

smartroad, 2015-07-27:
Would it be possible to add the ability to have two speeds for manual extrusion and retraction? I have to 'extrude' about 400mm of filament for my bowden and it is SLOWWWWW :laughing:

In Repetier Host it has the options to extrude fast or slow and retract fast only. This makes inserting the filament much quicker when feeding through a bowden setup.

foosel, 2015-07-27:
Sounds more like a job for custom controls to define something like an unload button.

spapadim, 2015-07-28:
Agree that a new feature is not necessary; e.g., here is the custom control I've been using for a while now (negative values for retracts):

controls:
- children:
  - commands:
    - G91
    - G1 E%(distance)s F800
    - G90
    input:
    - default: 200
      name: Distance
      parameter: distance
    name: Fast extrude
    type: parametric_commands
  name: Filament
  type: section

closed 2015-08-21 by foosel