OctoPrint/OctoPrint #996: [Plugin development] Sending custom gcode to printer without plugin hook

tmeghe, 2015-07-28:
Hi there,

I just read the documentation about plugin development, I managed to get mine working. I've a question about which way I have to choose to send custom GCODE command. GCODE plugin hooks "intercept" GCODE on the fly. My goal is to send custom code to the printer when I click on a button in the plugin settings tab.

I have a function in my __init__.py with my GCODE (actually the one from the example about rewriting M106 commands on the fly) but it's sent periodicaly to the printer. I want it to be sent only when I click a button in the settings tab.

What am I doing wrong ?

Thanks,
Thomas

closed 2015-07-28 by GitIssueBot

Salandora, 2015-07-28:
You could add a Javascript Asset, in that asset you can get the controlViewModel, in the controlViewModel is a function called sendCustomCommand. You could use this function example look at my EEPROM Repetier Plugin.

If you want to send a Command inside the init.py you can use "import octoprint.server" and then
octoprint.server.printer.commands(["M107 S0"])
as example ;-)

tmeghe, 2015-07-28:
Hi @Salandora thanks for your answer, I tried the second one (with the method provided by the server module) and it doesn't work.

I'll take a look at the controlViewModel method, but I'm very bad at JS...

Salandora, 2015-07-28:
Hmm this is strange the second method should have worked. But atlest the javascript variant should work. You can check my plugin for reference.

foosel, 2015-07-28:
@Salandora, close ;) Even better would be to just use the injected `_printer` instance (see [Injected Properties in the docs](http://docs.octoprint.org/en/master/plugins/concepts.html#injected-properties) - that's what it's for :) For the available methods on that, see the [PrinterInterface docs](http://docs.octoprint.org/en/master/modules/printer.html#octoprint.printer.PrinterInterface).

@tmeghe in general, "doesn't work" won't allow people to help you, "yields error x" on the other hand will. I also suggest to seek help/discussions about plugin development on the mailing list, the g+ community or the irc channel - the bug tracker is the wrong place for that (unless it's a request to add additional extension points within OctoPrint to solve a specific problem, see the point about Brainstorming tickets in the contribution guidelines linked by GitIssueBot above).

Salandora, 2015-07-28:
Shame on me, I wasn't aware of the _printer variable xD sorry

foosel, 2015-07-28:
Why do I even bother writing those docs? :P

Salandora, 2015-07-28:
Uhm for fun? :P