OctoPrint/OctoPrint #1001: "Connection" tab not unfolding

jminardi, 2015-07-30:
1. What were you doing?

Operating octoprint web interface

2. What did you expect to happen?

Connections tab to display when clicked

3. What happened instead?

Connections tab did not display when clicked

4. Branch & Commit or Version of OctoPrint:

Version: 1.2.4 (master branch)

5. Printer model & used firmware incl. version
   (if applicable - always include if unsure):

Marlin Integration branch

6. Browser and Version of Browser, Operating 
   System running Browser (if applicable - always 
   include if unsure):

Latest chrome on OS X

7. Link to octoprint.log on gist.github.com or pastebin.com
   (ALWAYS INCLUDE AND DO NOT TRUNCATE):

N/A

8. Link to contents of terminal tab or serial.log on 
   gist.github.com or pastebin.com (if applicable - always 
   include if unsure or reporting communication issues AND
   DO NOT TRUNCATE):

N/a

9. Link to contents of Javascript console in the browser 
   on gist.github.com or pastebin.com or alternatively a
   screenshot (if applicable - always include if unsure
   or reporting UI issues):

No errors seemed to show up:

```
Synchronous XMLHttpRequest on the main thread is deprecated because of its detrimental effects to the end user's experience. For more help, check http://xhr.spec.whatwg.org/.
packed_app.js?34dda18b:8354 Starting dependency resolution...
packed_app.js?34dda18b:8411 ... dependency resolution done
packed_app.js?34dda18b:8630 Going to bind 23 view models...
packed_app.js?34dda18b:8698 ... binding done
```

10. Screenshot(s) showing the problem (if applicable - always
    include if unsure or reporting UI issues):

I have read the FAQ.

ignaworn, 2015-07-30:
I would like to add that if you open the web  interface while the printer is desconected then the connection tab is expanded by default, and you can expand and contract normally.
Then if you connect to the printer and reload the webpage, the problem appears.

Using devel branch up to date and Google chrome on Windows 10.

markwal, 2015-07-30:
What plugins do you have installed?

markwal, 2015-07-30:
I thought maybe it was a side effect of something I did in ScreenSquish, but I've disabled it and still repro.  The reason it gets locked is that Collapse.transitioning gets stuck at 1 after #connection.collapse('hide') is called in the connection view model.  And it the Collapse.show it returns if transitioning is set.  I haven't figured out why transitioning gets stuck at true yet.

markwal, 2015-07-30:
It doesn't repro every time.  My current operating theory is if the connection state comes back before knockout propagates the isVisible to show the wrapper element (for logged in user) then the collapse gets hidden, but the transition doesn't occur (because the wrapper is hidden) and the transition completion event never fires leaving it permanently in the the transitioning state.

Seems like bootstrap has a bug in that a collapse of an element that is hidden by some container.

Not sure the best way to fix it. I could try to fix the bootstrap bug to call the completion function if the element is hidden for some other reason or we could postpone collapsing the connection state until the login state is known.

foosel, 2015-07-30:
PR merged to `devel` (thanks!), closing.

closed 2015-07-30 by foosel