OctoPrint/OctoPrint #1004: Improve serial conn startup for buggy firmware ?

spapadim, 2015-07-31:
1. What were you doing?

Connecting to a Solidoodle 2, after upgrading to the latest devel version.

[This is a resubmission of #1003 , to satisfy the bot; sorry :) ]

2. What did you expect to happen?

Open connection

3. What happened instead?

Serial connection stuck

4. Branch & Commit or Version of OctoPrint:

1.3.0.dev171+94625c4

5. Printer model & used firmware incl. version
   (if applicable - always include if unsure):

Solidoodle 2 - Marlin 1.0.0

6. Browser and Version of Browser, Operating 
   System running Browser (if applicable - always 
   include if unsure):

N/A

7. Link to octoprint.log on gist.github.com or pastebin.com
   (ALWAYS INCLUDE AND DO NOT TRUNCATE):

8. Link to contents of terminal tab or serial.log on 
   gist.github.com or pastebin.com (if applicable - always 
   include if unsure or reporting communication issues AND
   DO NOT TRUNCATE):

https://gist.github.com/spapadim/1c54db37601e4d291599

9. Link to contents of Javascript console in the browser 
   on gist.github.com or pastebin.com or alternatively a
   screenshot (if applicable - always include if unsure
   or reporting UI issues):

N/A

10. Screenshot(s) showing the problem (if applicable - always
    include if unsure or reporting UI issues):

Diagnosis and proposed "fix":

Upon upgrading an older printer (Solidoodle 2) from OctoPrint 1.0.0 to the most recent devel version, the printer connection got stuck immediately after opening.  Took me a while to figure out what's going on.  Turns out the firmware sends some `\0` characters upon startup (or at least the raspi receives them as such), so the "`ok`" for the first `M110` "hello" command does not appear at the beginning of a line, as `octoprint.util.comm` (line 887) expects it (see below).

As I understand, `_clear_to_send` implements a token bucket, which starts out empty (seems conservative?).  If that is so, then perhaps it could be initialized with >0 tokens (which is how I solved it, for now; just added a `_clear_to_send.set()` immediately after starting the serial queue thread) without any general ill effects (e.g., not certain if such a change would have any effects on serial line buffer over/underruns, under extreme circumstances)?  If that seems like a bad idea (ie should fix firmware bug instead), then feel free to just close ticket.Additional information

This is the Solidoodle start-up sequence:
```
2015-07-30 14:45:42,350 - SERIAL - DEBUG - Connecting to: /dev/ttyUSB0
2015-07-30 14:45:42,397 - SERIAL - DEBUG - Changing monitoring state from 'Offline' to 'Opening serial port'
2015-07-30 14:45:42,890 - SERIAL - DEBUG - Connected to: Serial<id=0x2a52690, open=True>(port='/dev/ttyUSB0', baudrate=250000, bytesize=8, parity='N', stopbits=1, timeout=30.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
2015-07-30 14:45:42,894 - SERIAL - DEBUG - Changing monitoring state from 'Opening serial port' to 'Connecting'
2015-07-30 14:45:50,278 - SERIAL - DEBUG - Recv: ^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@ [....more of the same]
2015-07-30 14:45:50,296 - SERIAL - DEBUG - Send: M110
2015-07-30 14:45:50,309 - SERIAL - DEBUG - Recv: echo:Marlin 1.0.0
2015-07-30 14:45:50,340 - SERIAL - DEBUG - Recv: echo: Last Updated: Dec  7 2013 23:40:35 | Autho^@r: Lawsy/Rincewind/Tealvince/Adrian
2015-07-30 14:45:50,353 - SERIAL - DEBUG - Recv: Compiled: Dec  7 2013
2015-07-30 14:45:50,376 - SERIAL - DEBUG - Recv: echo^@: Free Memory: 898  PlannerBufferBytes: 1232
2015-07-30 14:45:50,392 - SERIAL - DEBUG - Recv: echo:Stored setti^@ngs retrieved
2015-07-30 14:45:50,405 - SERIAL - DEBUG - Recv: echo:Steps per unit:
2015-07-30 14:45:50,424 - SERIAL - DEBUG - Recv: echo:  M92 X88.10 Y88.30 Z2^@267.72 E204.80
2015-07-30 14:45:50,440 - SERIAL - DEBUG - Recv: echo:Maximum feedrates (mm/s):
2015-07-30 14:45:50,454 - SERIAL - DEBUG - Recv: echo:  M203 X500^@.00 Y500
2015-07-30 14:45:50,481 - SERIAL - DEBUG - Recv: ^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@^@ok
2015-07-30 14:45:50,496 - SERIAL - DEBUG - Changing monitoring state from 'Connecting' to 'Operational'
[stuck]
```
compared to a Kossel (running Smoothieware), which works fine when _clear_to_send is initialized to zero:
```
2015-07-30 14:44:40,615 - SERIAL - DEBUG - Connecting to: /dev/ttyACM0
2015-07-30 14:44:40,652 - SERIAL - DEBUG - Changing monitoring state from 'Offline' to 'Opening serial port'
2015-07-30 14:44:41,144 - SERIAL - DEBUG - Connected to: Serial<id=0x1e3a3b0, open=True>(port='/dev/ttyACM0', baudrate=250000, bytesize=8, parity='N', stopbits=1, timeout=10.0, xonxoff=False, rtscts=False, dsrdtr=False), starting monitor
2015-07-30 14:44:41,147 - SERIAL - DEBUG - Changing monitoring state from 'Opening serial port' to 'Connecting'
2015-07-30 14:44:41,156 - SERIAL - DEBUG - Send: M110
2015-07-30 14:44:41,161 - SERIAL - DEBUG - Recv: ok
2015-07-30 14:44:41,173 - SERIAL - DEBUG - Changing monitoring state from 'Connecting' to 'Operational'
2015-07-30 14:44:41,179 - SERIAL - DEBUG - Send: M105
2015-07-30 14:44:41,674 - SERIAL - DEBUG - Recv: ok T:26.4 /0.0 @0 B:30.1 /0.0 @0
2015-07-30 14:44:41,679 - SERIAL - DEBUG - Send: M21
2015-07-30 14:44:41,692 - SERIAL - DEBUG - Recv: SD card ok
2015-07-30 14:44:42,162 - SERIAL - DEBUG - Recv: ok
2015-07-30 14:44:42,169 - SERIAL - DEBUG - Send: M20
2015-07-30 14:44:42,240 - SERIAL - DEBUG - Recv: Begin file list
2015-07-30 14:44:42,257 - SERIAL - DEBUG - Recv: config.txt
2015-07-30 14:44:42,277 - SERIAL - DEBUG - Recv: firmware.cur
2015-07-30 14:44:42,297 - SERIAL - DEBUG - Recv: config-override
2015-07-30 14:44:42,313 - SERIAL - DEBUG - Recv: End file list
2015-07-30 14:44:42,339 - SERIAL - DEBUG - Recv: ok
2015-07-30 14:44:46,186 - SERIAL - DEBUG - Send: M105
2015-07-30 14:44:46,213 - SERIAL - DEBUG - Recv: ok T:26.2 /0.0 @0 B:30.0 /0.0 @0
[...]
```

I have read the FAQ.

foosel, 2015-07-31:
No. If the firmware is buggy and sending garbage, then it's the firmware that should be fixed. There are already way too many workarounds for different firmware implementations in the code as is, adding more makes OctoPrint only harder and harder to maintain. These `\0` bytes are not part of the protocol. Might also be a problem with the printer's serial interface instead of the firmware, but this is not something that should be fixed in the host, it hints at a deeper problem and "fixing" it like that might actually only cover up other issues.

Feel free to write a plugin that does take care of that but no workarounds for that is going to be added to OctoPrints's core.

closed 2015-07-31 by foosel

spapadim, 2015-07-31:
Agreed (those nested if-else ladders in the monitor thread loop were no fun! :) ).

Plugin: hadn't thought of it, but I'll just keep an eye for conflicts on that single-line edit (unless, perhaps, if others find this issue).

Thanks again!