OctoPrint/OctoPrint #1005: Crash on Arch Linux

unlimitedbacon, 2015-07-31:
I get the following exception when I try to run Octoprint on Arch.

```
2015-07-31 11:51:26,125 - octoprint.server - ERROR - Uncaught exception
Traceback (most recent call last):
  File "run", line 22, in <module>
    octoprint.main()
  File "/home/matterhacker/OctoPrint/src/octoprint/__init__.py", line 81, in main
    octoprint.run()
  File "/home/matterhacker/OctoPrint/src/octoprint/server/__init__.py", line 394, in run
    self._server = util.tornado.CustomHTTPServer(self._tornado_app, max_body_sizes=max_body_sizes, default_max_body_size=s.getInt(["server", "maxSize"]))
  File "/usr/lib/python2.7/site-packages/tornado/util.py", line 215, in __new__
    instance.initialize(*args, **init_kwargs)
TypeError: initialize() got an unexpected keyword argument 'default_max_body_size'
```

Looks like something to do with Tornado. Installed version of python2-tornado is 4.2.1.

closed 2015-07-31 by GitIssueBot

foosel, 2015-07-31:
OctoPrint depends on Tornado 4.0.1

unlimitedbacon, 2015-07-31:
1. What were you doing?
  Trying to launch Octoprint

2. What did you expect to happen?
  I expected the server to start

3. What happened instead?
  It crashed and threw the following exception:
```
2015-07-31 11:51:26,125 - octoprint.server - ERROR - Uncaught exception
Traceback (most recent call last):
  File "run", line 22, in <module>
    octoprint.main()
  File "/home/matterhacker/OctoPrint/src/octoprint/__init__.py", line 81, in main
    octoprint.run()
  File "/home/matterhacker/OctoPrint/src/octoprint/server/__init__.py", line 394, in run
    self._server = util.tornado.CustomHTTPServer(self._tornado_app, max_body_sizes=max_body_sizes, default_max_body_size=s.getInt(["server", "maxSize"]))
  File "/usr/lib/python2.7/site-packages/tornado/util.py", line 215, in __new__
    instance.initialize(*args, **init_kwargs)
TypeError: initialize() got an unexpected keyword argument 'default_max_body_size'
```

4. Branch & Commit or Version of OctoPrint:
  master 9a6099ffc2982455d631c9d68a3273d9eb55885c

5. Printer model & used firmware incl. version
   (if applicable - always include if unsure):
  Not applicable

6. Browser and Version of Browser, Operating 
   System running Browser (if applicable - always 
   include if unsure):
  Not applicable

7. Link to octoprint.log on gist.github.com or pastebin.com
   (ALWAYS INCLUDE AND DO NOT TRUNCATE):
  https://gist.github.com/unlimitedbacon/51216ce73d25b2064836

8. Link to contents of terminal tab or serial.log on 
   gist.github.com or pastebin.com (if applicable - always 
   include if unsure or reporting communication issues AND
   DO NOT TRUNCATE):
 Not applicable

9. Link to contents of Javascript console in the browser 
   on gist.github.com or pastebin.com or alternatively a
   screenshot (if applicable - always include if unsure
   or reporting UI issues):
 Not applicable

10. Screenshot(s) showing the problem (if applicable - always
    include if unsure or reporting UI issues):
 Not applicable

I have read the FAQ.

unlimitedbacon, 2015-07-31:
Understood. Installing via setup.py instead of the [AUR package](https://aur.archlinux.org/packages/octoprint-devel-git/) takes care of this.

foosel, 2015-07-31:
As I said, OctoPrint depends on a different version of tornado, as specified in its `setup.py`. The new version is not API compatible, hence you are getting that error. Install 4.0.1 of tornado (use a virtual environment if you need the newer version for something else) and it will work.

unlimitedbacon, 2015-07-31:
Thanks. Is there any plane to move to the newer API?

foosel, 2015-08-01:
Yes, updating to current library versions is what I consider part of the
general maintenance, but as long as there are no pressing issues with the
existing dependency that has a quite low priority. I had to do some quite
heavy modding to get upload streaming to work (otherwise uploading 200mb
gcode files would cause low memory machines like the pi to get into
trouble), and therefore switching versions is not as simple as just
changing one or two lines here.

yhfudev, 2015-11-11:
What if I remove the arguments from the util.tornado.CustomHTTPServer ?
Is there any side effect? I know it's only a temporally solution.

https://aur.archlinux.org/cgit/aur.git/tree/octoprint-tornado401.patch?h=octoprint

foosel, 2015-11-11:
Yes, there is a side effect. At the very least uploads will not work as intended with regards to size restrictions. More problems might be down the road too.

cbxbiker61, 2016-09-29:
foosel,

I understand your desire to maintain as robust a system as possible.  But let me say that in my case, I just relax the restrictions on the versioning, i.e. remove the "<" side of the requirements, and all seems to be good, with the exception of Tornado, I'm running the dev versions out of git.

Is there a way that you can think of that would allow those of us with coding skills give you feedback as to how the latest versions of python modules are working with Octoprint so that your dependencies don't stagnate?  What we need to consider is that there may be tangible improvements in the dependent modules that Octoprint may benefit from.  We assume that the module maintainers are striving to move forward with improvements, but no one is actually going to be able to vet the improvements in each module line by line.

FirefighterBlu3, 2016-12-29:
@unlimitedbacon remove the `default_max_body_size` keyword and rename all instances of `max_body_sizes` to `max_body_size`