OctoPrint/OctoPrint #1007: "Print" button enabled on connection, before file is selected.

AndKe, 2015-08-03:
1. What were you doing?
Connecting to the printer

2. What did you expect to happen?
the WEBUI to give relevant options.

3. What happened instead?
the WEB UI enabled "print" button on connection to printer, before anything to print is selected.

4. Branch & Commit or Version of OctoPrint:
1.2.4 master

5. Printer model & used firmware incl. version
hommade big prusa mendel lookalike, with marlin firmware

6. Browser and Version of Browser, Operating 
Ubuntu 15.04 + Firefox

7. Link to octoprint.log on gist.github.com or pastebin.com
no link really needed, just connect to a printer and observe "print" button being enabled

8. Link to contents of terminal tab or serial.log on 
no link really needed, just connect to a printer and observe "print" button being enabled

Or, told in a simple way:

Once the printer is connected, the "print" button is enabled, and it have no function.
This is the state after connection: (and "print" should be disabled until something is selected)
Machine State: Operational
File:  
Timelapse: -
Approx. Total Print Time: -
Print Time: -
Print Time Left: -
Printed: -

foosel, 2015-08-03:
There is a template for a reason... ;)

AndKe, 2015-08-03:
oh, I thought nobody would ever see this ..  :)   The nature if the issue is just too simple, too easy to reproduce to fill out a template :)
And - let's face it, the software works great as-is , this is more about polishing the masterpiece.

foosel, 2015-08-03:
The template is necessary for most issues however (since otherwise I spend more time running after the information necessary to debug a problem than actually debugging and fixing it), so it's required for all bug reports. Sorry, I wish it wasn't necessary either, but before the template and the bot there were times I literally could not get any work done due to hunting down information.

ntoff, 2015-09-05:
oh hey it's there in Version: 1.2.7-dev-1-ga2e5fc0-dirty (maintenance branch) too

I'd never even noticed it until now.

foosel, 2016-01-25:
Well, now it's not doing that anymore. Fixed in `maintenance` and `devel` branch, will be included in 1.2.9 release.

commit bcd7bb4d (2016-01-25): Don't enable Print button if no job is selected

Fixes #1007
closed 2016-01-25 by foosel