OctoPrint/OctoPrint #1008: octopi.local problem on Win 10

Mdonci, 2015-08-03:
Hi,

Since I upgraded from 8.1 to 10 I am unable to connect to my Pi. Tried to reboot the pi, my computer as well. Several times, didn't work. 
Now I disabled the firewall and Avast, but it didn't resolve the problem.
I get the following error message all the time:
ERR_NAME_NOT_RESOLVED
Can anybody help me resolve this issu?
Thank!

closed 2015-08-03 by GitIssueBot

markwal, 2015-08-03:
I'm running 10 and I don't have this difficulty, but then I'm running a dns server on another pi in the house so it can resolve the name.

My guess is that perhaps you were using lmhosts to point at your pi and upgrading to Win 10 lost your lmhosts file?  You could create another one and see?  Probably you should post this question to the google group since there are more people watching that one and perhaps someone also hit this issue.

https://groups.google.com/forum/#!forum/octoprint

foosel, 2015-08-04:
Might be that there's a problem with bonjour on Windows 10. In any case, not an OctoPrint problem, not even an OctoPi problem but rather a Windows 10 problem, so this bug tracker (and the one of OctoPi) are the wrong place and the Mailinglist/Google Group is more appropriate.

ignaworn, 2015-08-04:
You can do, as a temporary workaround, the following.
Open command line and type
ipconfig /flushdns and try again