OctoPrint/OctoPrint #1009: OctoPrint stalls and disconnect on SDInit Fail

pixel3design-hub, 2015-08-04:
I use OctoPi for a long time, but updating to a MarlinKimbra FW makes it disconnect because a non working anymore SD Reader on the REPRAP_DISCOUNT_SMART_CONTROLLER.

There should be a way to disable the SD check or Octoprint will become unusable to me.

closed 2015-08-04 by GitIssueBot

markwal, 2015-08-05:
OctoPrint should work even when an SD card is not present.  It sends an M21 to initialize it, but it will just set it unavailable if it gets a failure back.  Are you saying that sending an M21 to your printer makes it disconnect itself?  A bit of your serial.log would be useful for diagnosing here, but since you may not have a bug here you might want to start by posting a question to the Google group.

Salandora, 2015-08-07:
Please provide a serial.log for this maybe the MarlinKimbra firmware send something like Error: SDInit failed. Which would explain why OctoPrint gets disconnected because on every Error response it is supposed to disconnect.