OctoPrint/OctoPrint #1010: Nozzle clean step removal

Hiltonny, 2015-08-04:
There should be setup choice to allow or not allow the clean step when the print head goes from the home position to the warmup position as a print begins. The current software moves the print nozzle to the edge of the glass build platform so any hanging plastic can be sheared off. This clean step places un-needed stresses on the printer unless the setup is perfect because the nozzle has to deflect as it hits the glass edge. After the cold print head reaches the warmup position more plastic can exit the nozzle as it warms up to temperature anyway. The clean step can also throw off the build platform leveling adjustments.

closed 2015-08-04 by GitIssueBot

foosel, 2015-08-04:
Whatever you are observing, it is from your own gcode files that you are printing. OctoPrint doesn't move the print head anywhere on its own upon a print start. Fix the start gcode in your slicer ;)

nophead, 2015-08-04:
See http://hydraraptor.blogspot.co.uk/2012/04/ooze-free-unattended-start.html for a better start procedure. Works fine in OctoPrint, or any other host or standalone.

Hiltonny, 2015-08-04:
This means Type A machines added this feature?

foosel, 2015-08-04:
I don't know, ask them. it's either that or it's in your slicer's start gcode.

In any case, if you are running a fork, please report your issues with it to the people maintaining the fork.