OctoPrint/OctoPrint #1011: [request] multiple-objects printing

Renha, 2015-08-05:
Add support for per-model settings with multiple-objects printing, which recently has been added to CuraEngine.

Probably we need new dialog "multiple object printing" with list of
"modelfile   ---   objectsettings" (I am not sure we need to use whole Cura-profile files there)

Maybe not new dialog, but update old "Select slicing settings" dialog.

foosel, 2015-08-05:
That is in fact something on my personal wish list too, but first a couple of other things would need to be put into place:

  * slicing profile editor accessible from slicer dialog
  * visual editor to control placement, scaling and rotation of one model accessible from slicer dialog
  * adjustment of internal slicing API to allow scaling & rotation
  * adjustment of internal slicing API to allow more than one model per plate
  * some ui metaphor to allow plating more than one model
  * adjusting existing slicing plugins to allow multiple models

I don't think this will all make it into 1.3, so be aware that while this is something I'd like to see, it's still far off.

Renha, 2015-08-05:
I see. What are you thinking about using OpenSCAD as backend for operations like rotation, placement, visualisation? It could generate pictures, and do everything else. It should not be too hard to make a basic plugin, maybe even I could do such a thing, but making user-friendly interface would be harder.

foosel, 2015-08-05:
I was rather thinking along the lines of keeping the actual means how to do that strictly within the individual slicing plugin. So providing the necessary transformation matrix to that, as gained from the user input in the visualizer, and then leaving it up to the plugin on how to achieve the actual transformation for the provided stl file.

techied, 2016-01-15:
I rather like this idea, I have been planning to print some objects that require 5 of the same thing, and I wish I could just enter a qty: 5 and have it auto-arrange and print them (No visualization needed)

Renha, 2016-02-17:
hi @foosel ! how's your timeline now for that feature? sorry for annoying, just to know for better planning of building my printers

curaengine guys are preparing their release 2.1 now, we may switch to it without multiple object printing for now, to make that task easier to do later

foosel, 2016-02-17:
No time

kennethjiang, 2016-09-28:
Just released a GUI-based slicer plugin: https://github.com/kennethjiang/OctoPrint-Slicer

With some enhancement it can be used to do multi-object printing. If any of you can pitch in to share a bit of the dev work, please let me know.

Salandora, 2016-09-28:
Woah awesome @kennethjiang
This is definately a plugin I need to install when home. Good job.