OctoPrint/OctoPrint #1019: Internal serer error after reboot

guysoft, 2015-08-09:
1. What were you doing?
I had a thermal runaway and I rebooted the octoprint server
2. What did you expect to happen?

3. What happened instead?
I am getting internal server error at the webpage

4. Branch & Commit or Version of OctoPrint:
1.2.4 latest from OctoPi 0.12 update

5. Printer model & used firmware incl. version
   (if applicable - always include if unsure):
Rigidbot  Marlin 1.0.2

6. Browser and Version of Browser, Operating 
   System running Browser (if applicable - always 
   include if unsure):
Not relevant

7. Link to octoprint.log on gist.github.com or pastebin.com
   (ALWAYS INCLUDE AND DO NOT TRUNCATE):

8. Link to contents of terminal tab or serial.log on 
   gist.github.com or pastebin.com (if applicable - always 
   include if unsure or reporting communication issues AND
   DO NOT TRUNCATE):

```
2015-08-09 17:09:11,251 - octoprint.server.util.flask.fix_webassets_filtertool - ERROR - Got an exception while trying to apply filter, ignoring file
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.2.4-py2.7.egg/octoprint/server/util/flask.py", line 198, in fixed_wrap_cache
    self.cache.set(key, content)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.2.4-py2.7.egg/octoprint/server/util/flask.py", line 135, in fixed_set
    dir=self.directory)
  File "/usr/lib/python2.7/tempfile.py", line 300, in mkstemp
    return _mkstemp_inner(dir, prefix, suffix, flags)
  File "/usr/lib/python2.7/tempfile.py", line 235, in _mkstemp_inner
    fd = _os.open(file, flags, 0600)
OSError: [Errno 2] No such file or directory: '/home/pi/.octoprint/generated/.webassets-cache/.aae992ee2e995b3bbbdbcfe1d3a4d868_3ZiRy'
2015-08-09 17:09:11,263 - octoprint - ERROR - Exception on / [GET]
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1817, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1477, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1381, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1475, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1461, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.2.4-py2.7.egg/octoprint/server/util/flask.py", line 274, in decorated_function
    rv = f(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.2.4-py2.7.egg/octoprint/server/views.py", line 274, in index
    **render_kwargs
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/templating.py", line 128, in render_template
    context, ctx.app)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/templating.py", line 110, in _render
    rv = template.render(context)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Jinja2-2.7.3-py2.7.egg/jinja2/environment.py", line 969, in render
    return self.environment.handle_exception(exc_info, True)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Jinja2-2.7.3-py2.7.egg/jinja2/environment.py", line 742, in handle_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.2.4-py2.7.egg/octoprint/templates/index.jinja2", line 14, in top-level template code
    {% include 'stylesheets.jinja2' %}
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.2.4-py2.7.egg/octoprint/templates/stylesheets.jinja2", line 9, in top-level template code
    {% assets "less_app" %}
  File "/home/pi/oprint/local/lib/python2.7/site-packages/webassets-0.10.1-py2.7.egg/webassets/ext/jinja2.py", line 186, in _render_assets
    urls = bundle.urls()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/webassets-0.10.1-py2.7.egg/webassets/bundle.py", line 787, in urls
    urls.extend(bundle._urls(new_ctx, extra_filters, *args, **kwargs))
  File "/home/pi/oprint/local/lib/python2.7/site-packages/webassets-0.10.1-py2.7.egg/webassets/bundle.py", line 746, in _urls
    *args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/webassets-0.10.1-py2.7.egg/webassets/bundle.py", line 625, in _build
    hunk.save(output_filename)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/webassets-0.10.1-py2.7.egg/webassets/merge.py", line 164, in save
    f.write(self.data())
TypeError: must be unicode, not str
```

9. Link to contents of Javascript console in the browser 
   on gist.github.com or pastebin.com or alternatively a
   screenshot (if applicable - always include if unsure
   or reporting UI issues):
Nope

10. Screenshot(s) showing the problem (if applicable - always
    include if unsure or reporting UI issues):
Its a boring picture

I have read the FAQ.

foosel, 2015-08-09:
Ok, I thought I'd fixed this (it happened a couple of times during development but is very hard to reproduce), might be something new though too, hard to tell from the *truncated* log.

Try `rm -r /home/pi/.octoprint/generated`, then restarting the server, that should hopefully fix it, if not please report back. Also in any case please provide the startup log entries that include the plugin list (those are the reason for the big "DO NOT TRUNCATE" there, among other things) - I see it's happening upon rendering out the less bundle, so it might also be caused by some plugin too.

If that doesn't help, you can disable the use of the webassets stuff by setting `devel.webassets.bundle` to `false` in your config.yaml to hopefully get your server starting again, but please report back the above stuff in any case.

guysoft, 2015-08-09:
A reboot of of the pi seemed to fix it.
Full log here:
http://gnet.homelinux.com/files/octoprint.log

foosel, 2015-08-11:
I think the cause for this was both a lack of error resilience within the cache implementation of webassets and a possible timing issue when deleting the folder on server startup. I pushed 38be47c (and 0c44762f6fbf88f8f205032182f743e8fb937dc1) plus the above commits that should monkey patch webassets to be more forgiving when the cache cannot be read or written (that should then clear itself up further down the road in such situations I hope) and also made the folder reset a bit more intelligent (and resilient against win32 shenanigans I discovered while doing that).

Is on `devel`, will also be released with 1.2.5.

I hope that takes care of this.

closed 2015-08-11 by foosel