OctoPrint/OctoPrint #1024: Multible Webcams in Octoprint

mberger1982, 2015-08-14:
Hi
is it possible to build in a dual-Webcam view in the octoprint application?

mberger1982, 2015-08-14:
I'm sorry...
well i just wanted to ask, if it is possible, to add a Dual-Webcam function in the octoprint...
now there is in the register "control" only one webcam supported, it would be great, if there could be added a secound one.
i use a reprap x350 with 2 webcams, one is in the octoprint window and the secound in a seperate browser-window made with mjpg-stream and livestream.sh (http://jacobsalmela.com/raspberry-pi-webcam-using-mjpg-streamer-over-internet/) and it is a little bit boring to open 2 browserwindows to see the both cams...

thank you!

foosel, 2015-08-14:
Duplicate of issue #401 (I admit the search is sometimes tricky to get to behave, but this time a simple search for the term "webcam" was all that it took ;))

closed 2015-08-14 by foosel

mberger1982, 2015-08-14:
ok thanks