OctoPrint/OctoPrint #1030: [Request] Warn user about disk space issues

foosel, 2015-08-20:
OctoPrint should display a big red warning notification when the disk space for file uploads starts becoming a rare good (compare also #655). It should also detect when the disk space on the partition the configuration file is stored on runs low and refuse to even save settings from a certain point onward to prevent corruption of files.

- [ ] add setting for warning and "refuses to operate" space thresholds
- [ ] add new API to provide data (free vs total) for all the configurable folders (uploads, timelapses, timelapse temp, logs, watched, plugin data folder, ...) plus the location of the configuration file
- [ ] utilze API from frontend to query data and display information next to folder paths in the settings
- [ ] also display warning if user is admin when data anywhere falls below a certain threshold, include information how to free up space
- [ ] optional: include new hook for plugins to provide additional space information

ntoff, 2015-08-29:
Also what about temp space for mjpg streamer? I know it's not a part of octoprint but how would you handle things if say, you had 1gb free space and mjpg streamer took all of it with its temp files? Would you also warn users that there's potentially low space issues if they happen to be using a webcam and the print will take over an estimated certain amount of time (potentially using up all that temp space)?

foosel, 2015-08-31:
> I know it's not a part of octoprint

Actually it is. The capturing for timelapse snapshots is done by OctoPrint, the only part the mjpg_streamer plays here is providing the snapshot URL from which the frames are captured.

However, the temporary folder is part of the "all configurable folders" part above.

ntoff, 2015-08-31:
Sorry, what I meant by the second question was does octoprint know how much space is theoretically required for a time lapse video + snapshots based on how long it thinks the print will take? And does it take that into account when starting a print (for the "refuse to operate" feature)?

So if I set a warning for 500mb because gcode files are small, but I start a long print and there's say, 1gb left on the disk then octoprint will be happy. But while printing what if the time lapse ends up requiring 2gb of space.

Is there any way for octoprint to look at the remaining space and the timelapse quality settings and say "ok, this print will go for A hours, each snapshot will theoretically be 300kb, I'm going to need X amount of space for it and Y amount for the end video" and possibly throw a warning that, while there's sufficient space now, you will probably run out half way through the print and end up with a truncated timelapse video?

foosel, 2015-08-31:
Hm... Only by adding a setting of the average snapshot size. That depends on too many factors (resolution, quality, compressability which depends on the image itself etc) to reliably guess.

ntoff, 2015-08-31:
Ok so what about a general sort of informational only notice in the timelapse settings that based on the settings currently in use (for timed mode), a 1 hour print with an average file size of 300kb will take X amount of images and Y amount of space and for layer change mode an item with 500 layers will take X images and Y space. Maybe make the average file size configurable and let us end users just figure out our own best setting based on our own testing we can do?

If my coding ability wasn't limited to copy and paste I'd have a go at it myself but all I can do is help test things. :(

I wonder how hard it would be to figure out the last average size and over time that value would be used?

foosel, 2015-08-31:
The problem with things like "last average size" is that you can go completely wrong when the user changes the resolution, the webcam angle or anything else like that. That makes it risky to depend on something like that.

An informational only notice kind of thing would be ok this way ("Based on your last timelapses, this one will take approximately X MB per hour printing time or Y MB per 500 layers"). That gives you at least the temporary space. Then you run into the next problem though when trying to guess the file size of the final rendered movie - you'll need the accumulated size of all temporary files plus the final rendered movie plus some safety margin in free space, otherwise stuff will break.

To be honest, I'm not so sure it is worth the effort sinking too much time into prediction stuff like that. We can't take every responsibility off the user's hands through automation, guesswork and all that. When the disk is full it is full. A simple approach would be "when warning threshold is reached, always warn when starting a print with a timelapse". Another option still would be to allow the definition of multiple threshold values (since the various folders in fact can be located on different partitions, nobody stops you for example from attaching an external USB drive with hundreds of GB to use as your timelapse folder). But trying to guess what is a problem and what isn't? Bad cost/usefulness ratio ;)

ntoff, 2015-08-31:
Or here's an idea, why not just a 3rd configurable warning level? If timelapse is enabled warn at this space free, if not then warn at this lower space. That might be the easiest and then like you say we end users can take responsibility for how much space we think we'll need.

HarlemSquirrel, 2018-11-04:
Has anyone taken a stab at this yet? I'm a Ruby dev who has dabbled in Python. I'd be willing to give it a go.

foosel, 2018-11-05:
@HarlemSquirrel not that I know of, so feel free! Shall I mark it as grabbed?

HarlemSquirrel, 2018-11-06:
Sure! I'll take a swing at it :grinning:

foosel, 2018-11-06:
Marked as grabbed by @HarlemSquirrel

HarlemSquirrel, 2018-11-11:
@foosel it looks like much of the original requirements have been implemented. The files panel shows free vs total disk space. A warning shows when the free space falls below the set warning level and then the text becomes red when the free space falls below the critical threshold. 

Could we please confirm the outstanding requirements?

- utilze API from frontend to query data and display information next to folder paths in the settings
- also display warning if user is admin when data anywhere falls below a certain threshold, include information how to free up space
- optional: include new hook for plugins to provide additional space information

foosel, 2018-11-12:
> The files panel shows free vs total disk space. A warning shows when the free space falls below the set warning level and then the text becomes red when the free space falls below the critical threshold.

Only for the uploads folder though, and not very prominently for the red range. 

All the other (configurable) folders are still lacking, e.g. folder the configuration resides in, plugins data folder, timelapse folder, ...

And not very prominently. The idea is here to add a new overlay (like the "server not available" or the "please reload" one) if available space in config/plugin data becomes critical (and any folder if the user is an admin) and add an api for the frontend to query and show the current "fill level" in the folder config settings screen.

I've hopefully clarified the initial post a bit.