OctoPrint/OctoPrint #1033: Sending:M105 and MAXTEMP Error

Shieldudaram, 2015-08-24:
I'm new to this but i have the prusia i3 from folgertech fully built now but when i hit connect I get this....

start
Printer is now online.
echo:Marlin1.0.2
echo: Last Updated: Aug 24 2015 01:55:55 | Author: (none, default config)
Compiled: Aug 24 2015
echo: Free Memory: 5238  PlannerBufferBytes: 1232
echo:Hardcoded Default Settings Loaded
echo:Steps per unit:
echo:  M92 X78.74 Y78.74 Z533.33 E836.00
echo:Maximum feedrates (mm/s):
echo:  M203 X500.00 Y500.00 Z5.00 E25.00
echo:Maximum Acceleration (mm/s2):
echo:  M201 X9000 Y9000 Z100 E10000
echo:Acceleration: S=acceleration, T=retract acceleration
echo:  M204 S3000.00 T3000.00
echo:Advanced variables: S=Min feedrate (mm/s), T=Min travel feedrate (mm/s), B=minimum segment time (ms), X=maximum XY jerk (mm/s),  Z=maximum Z jerk (mm/s),  E=maximum E jerk (mm/s)
echo:  M205 S0.00 T0.00 B20000 X20.00 Z0.40 E5.00
echo:Home offset (mm):
echo:  M206 X0.00 Y0.00 Z0.00
echo:PID settings:
echo:   M301 P22.20 I1.08 D114.00
Error:0
: Extruder switched off. MAXTEMP triggered !
Error:Printer stopped due to errors. Fix the error and use M999 to restart. (Temperature is reset. Set it after restarting)
SENDING:M105

the sending:m105 was from me manually trying to check the temp and it never does anything after that. Any help would be very greatly appreciated.

nophead, 2015-08-24:
Nothing to do with Octoprint. Most likely your extruder thermistor is
shorted.

Shieldudaram, 2015-08-24:
How could I fix this? It doesn't appear to be shorting itself anywhere.

nophead, 2015-08-24:
Use a multimeter to measure the resistance. Should be about 100K at room temperature. Get support from folgertech, not here. Octoprint is just showing the error messages coming from your firmware.

foosel, 2015-08-25:
Yes, this is the wrong place, we can't help you here. It's a hardware problem with your printer, you'd get the same message with any other host.

closed 2015-08-25 by foosel