OctoPrint/OctoPrint #1036: no attribute _basefolder

punkkeks, 2015-08-30:
> AttributeError: 'PiAPl' object has no attribute '_basefolder'

when I try to register an error_handler via `self.get_blueprint().app_errorhandler(500)(self.api_errorHandler)` the following exception occurs

>Traceback (most recent call last):
>  File "/home/marcel/build/OctoPrint/src/octoprint/plugin/core.py", line 660, in load_plugin
>    plugin.load()
>  File "/home/marcel/studium/bachelor-thesis/piapl/octoprint_PiAPl/__init__.py", line 8, in __plugin_load__
>    p = PiAPl()
>  File "/home/marcel/studium/bachelor-thesis/piapl/octoprint_PiAPl/piapl.py", line 30, in __init__
>    self.get_blueprint().app_errorhandler(500)(self.api_errorHandler)
>  File "/home/marcel/build/OctoPrint/src/octoprint/plugin/types.py", line 887, in get_blueprint
>    kwargs = self.get_blueprint_kwargs()
>  File "/home/marcel/build/OctoPrint/src/octoprint/plugin/types.py", line 909, in get_blueprint_kwargs
>    static_folder = self.get_asset_folder()
>  File "/home/marcel/build/OctoPrint/src/octoprint/plugin/types.py", line 162, in get_asset_folder
>    return os.path.join(self._basefolder, "static")
>AttributeError: 'PiAPl' object has no attribute '_basefolder'

punkkeks, 2015-08-30:
Update:
When I fix the above error message, the following will occurr

>Traceback (most recent call last):
>  File "/home/marcel/build/OctoPrint/src/octoprint/plugin/core.py", line 660, in load_plugin
>    plugin.load()
>  File "/home/marcel/studium/bachelor-thesis/piapl/octoprint_PiAPl/__init__.py", line 8, in __plugin_load__
>    p = PiAPl()
>  File "/home/marcel/studium/bachelor-thesis/piapl/octoprint_PiAPl/piapl.py", line 32, in __init__
>    self.get_blueprint().app_errorhandler(500)(self.api_errorHandler)
>  File "/home/marcel/build/OctoPrint/src/octoprint/plugin/types.py", line 888, in get_blueprint
>    blueprint = flask.Blueprint("plugin." + self._identifier, self._identifier, **kwargs)
>AttributeError: 'PiAPl' object has no attribute '_identifier'
>

what is self._identifier supposed to be?

punkkeks, 2015-08-30:
fixed that issue too by setting _identifier to PiAPl. Now they all are gone, but fun fact: the error_handler does not work :D what is the proposed way to register it? I thought it should go like that.

foosel, 2015-08-30:
This is not an issue with OctoPrint but how you are writing your plugin. The mailing list or the G+ community, or even the IRC channel, are better places for these kinds of questions.

OctoPrint's plugin system injects a bunch of properties into your plugin implementation (as described in the documentation). It does this *after* you have constructed your plugin object, so after `__init__`. For this reason it defines an `initialize` method for your plugin to do its setup at a point in time when all properties have been injected.

See http://docs.octoprint.org/en/master/modules/plugin.html#octoprint.plugin.core.Plugin and http://docs.octoprint.org/en/master/plugins/concepts.html#lifecycle

closed 2015-08-30 by foosel

foosel, 2015-08-30:
To elaborate a bit further maybe... Your plugin is trying to do something with the blueprint defined by the BlueprintPlugin mix in, but it does that *before* the plugin system has had a chance to initialize your plugin properly, which leads to the failures you are seeing. Anything that is not just declaring variables and doing things that do not *in any way* depend on OctoPrint should be done in the `initialize` method for that reason, since the plugin system will call that method when it's done setting up everything, making it basically what should be your plugin's starting point. 

Depending on what you are trying to do however something like a StartupPlugin mix in might even be better suited. It is impossible to determine your goals however from two stack traces and a somewhat ambiguous plugin name ;)

punkkeks, 2015-08-30:
Thanks. Your comment helped me. I tried to call the blueprint thing in the __init__ function and thatswhy nothing is initialized. I moved it up to the initialize() function and now it works like a charm ;) Sorry for the noise