OctoPrint/OctoPrint #1037: Timelapse start cleans tmp directory before avconv is finished

JoseAlcerreca, 2015-08-31:
The tmp directory is cleared when a new timelapse starts:

```
def startTimelapse(self, gcodeFile):
		self._logger.debug("Starting timelapse for %s" % gcodeFile)
		self.cleanCaptureDir()
...
```

but it's not checking if the previous timelapse is already generated. This is a problem if the cpu is slow (RPi) and you start a new print immediately after finishing the previous one. I've lost hours of video.

My suggestion is to use `tmp/{filename}/{n}.jpg` instead of `tmp/{n}.jpg` or something similar and maybe implementing some kind of processing queue.

ntoff, 2015-08-31:
I wonder if that's why the end of one of my videos got cut off, although I didn't start a new print right away.

foosel, 2015-09-11:
Duplicate of #318 

(Search: ["timelapse"](https://github.com/foosel/OctoPrint/issues?utf8=%E2%9C%93&q=is%3Aissue+is%3Aopen+timelapse))

closed 2015-09-11 by foosel