OctoPrint/OctoPrint #1040: [request] Send to SD Card from Local File

mikenew12, 2015-09-01:
It would be really nice to be able to initiate a transfer to the SD card from a locally stored file. That way transfers could be done during downtime or after a print performs poorly due to the USB connection limits.

I know the little button menu is getting pretty crowded, but maybe there could be a little popup to hold extra features? Other useful things could maybe be there, like renaming files (which would also be nice). Maybe something like this:
[image]

Salandora, 2015-09-01:
I had once done a whole FileManager for OctoPrint, which should get this feature too.
My guess is it would be extremely unlikely to stuff more things into that little menu.

If I find a bit of time I will convert my FileManager Branch into a Plugin.
And add this Feature to it.
But this could take a while 14-21 Days before I can start it.

mikenew12, 2015-09-02:
That would be really great actually. A full blown file manager sounds like a pretty major feature. Let me know if I can help in any way.

I don't think the little context menu is a bad option though. Even with a file manager, it would still be convenient to trigger some operations directly from the main file viewer section, and having a context menu is pretty normal for something like that.

ntoff, 2015-09-04:
What about something like an ajax file explorer? I used one on my NAS a while ago. Something that runs outside of Octoprint, but gives a full browser based file manager.

Salandora, 2015-09-19:
I started working on the fileManager a few days ago, but to get it working correctly, I first need to get OctoPrint's folder support (with all the necessary commands) done.

foosel, 2017-01-27:
Filemanager plugin is released, OctoPrint 1.3.0 with folder support is as well, considering this closed.

closed 2017-01-27 by foosel