OctoPrint/OctoPrint #1045: estimated printing time not calculated in border case

ss89, 2015-09-02:
The estimated printing time calculation seems to be aborted when loading a huge gcode file and pressing print before the calculation is done.
I experienced this issue on commit 8fddd8dd87d8c6deb991758e15f1d985cdb97feb (devel).

ss89, 2015-09-02:
octoprint.log: https://gist.github.com/ss89/5936db52620b4fe91cd5

foosel, 2015-09-02:
That is intentional behaviour. The estimation based on the GCODE file costs a lot of CPU due to the calculations involved and hence will be aborted when a print job is started in order to not negatively influence the print result. It will be taken up again when the print job finishes. What is displayed under "Approx. Total Print Time" (or "Ungefähre Druckdauer" in your case) is the result from that estimation. If you hit print before that estimate is ready, you'll not have a value there (since there is no estimate).

closed 2015-09-02 by foosel