OctoPrint/OctoPrint #1047: Rotating 90 degree webcam not working for iOS browser

javichumellamo, 2015-09-04:
I've tested the rotate 90 function either from an IPad, an iPhone and a Mac desktop and it only works for the desktop computer. I've tried disabling and re-enabling this feature but this behaviour remains. From an android device it also works fine

foosel, 2015-09-05:
@javichumellamo please state exact browser (version etc). Also exact version number of OctoPrint, otherwise we don't have the information necessary to debug this properly, as you already got explained above.

javichumellamo, 2015-09-05:
I've tested it on iOS 8.4 and 8.4.1 with Safari browser. Octoprint is running 1.2.6. I've installed it from the octopi image and upgraded to the last available version

markwal, 2015-09-07:
Fixed in devel branch (almost, it will be the next time the css gets regenerated and committed).

closed 2015-09-07 by markwal

foosel, 2015-09-07:
Don't forget the change log entry :)