OctoPrint/OctoPrint #1048: How long did it took to be printed in file details

javichumellamo, 2015-09-04:
In the Files list, you can see when a file was printed for the last time, it will be quite useful to show how long did it took.

foosel, 2015-09-05:
Should only be done for the last print time on the current printer profile.

@javichumellamo I've now manually marked this as a request but _please_ read what the bot told you above, you could have done that yourself by just prefixing the title of you ticket and that way already helped us help you.

ntoff, 2015-09-05:
Am I missing something? It does show that stuff, in the history tab.

markwal, 2015-09-05:
Yes, @javichumellamo, the duration of the last print shows when you open the info for an item with the down arrow button.  Also, there is the PrintHistory plugin which you might find useful.

javichumellamo, 2015-09-05:
@markwal, when I open the info for an item, I can see the estimated print time, but what I'm wanting to get is the real time it took to be printed for the last time. I know I can use PrintHistory, but I think this simple change will improve the UI usability.

ntoff, 2015-09-05:
so, like this?
http://i.imgur.com/26GM8XU.jpg

javichumellamo, 2015-09-05:
That's !!! it seems to work for me running 1.2.6 stable but I agree, we need additional testing

ntoff, 2015-09-05:
You can also leave the if condition in there if you change it too but I removed it because with it there, if there's no data then it just doesn't show up and for me, I would rather see it just give me a dash (-) if there's no data, than hide the information all together.

That's just personal preference though and leaving the if condition in there, makes it look more clean I suppose.

hgeorge94, 2016-02-22:
It would appear that this issue might already be implemented, but it is still up-for-grabs. Does this issue need further work, or does it just need to be resolved?

foosel, 2016-02-22:
Never got a PR for that, so it is still open.

foosel, 2016-10-04:
Now I got a PR for that by @i-am-me which is now merged on the `devel` branch, hence this can be considered implemented. Will be part of the 1.3.0 release :)

closed 2016-10-04 by foosel