OctoPrint/OctoPrint #1049: [request] cancel without error/fail status

ntoff, 2015-09-05:
Hitting the cancel print button also seems to set the status to a failed print. This might not always be the case. Sometimes I'll only bother to half print a test calibration piece, or it might be that I have to leave and so I might cancel a print if it's only just started. Also I've sometimes realised after hitting print that I've used the wrong gcode (I have 2 printers which use different startup gcode files) so have to hit cancel to reupload.

It would be nice if those conditions didn't trigger a "failed" status on a print since the print didn't fail, I just didn't wish to print that item right now.

I just hate seeing "failed" prints in the history :(

Is it possible to somehow have maybe a way to press a modifier key like shift or ctrl and press cancel and it would then cancel the print without setting the failed state?

foosel, 2015-09-06:
Don't take this the wrong way but don't you think it a bit of a skewed cost/usefulness ratio to introduce modifier keys for canceling prints just to not have red file names in your file list? ;)

If you cancel a print it obviously didn't run from start to finish. So any data related to that print (like it's runtime for example) are not applicable to fill prints. Hence it gets marked as failed when you cancel it. Just adding a modifier key wouldn't be enough here, the API would need to be adjusted, the data tracking would need to be adjusted, etc. Just so that there are less red lines in that corner case.

Salandora, 2015-09-07:
To be honest, I would like to have some feature like this too xD
I also don't like the red lines when I hit cancel because something is wrong before the print starts.

So If nobody is against it I would put it onto my todo list :-)

foosel, 2015-09-07:
If you want to look into this be my guest, but keep in mind that you need to stay backwards compatible with regards to the existing data and API. 

I still think that if you guys hate the red lines so much (this is the view problem if I understand it correctly) maybe it would make more sense to create a plugin that simply gets rid of those altogether instead. That would be a minimally invasive thing to do and leave it up to the individual user.

Salandora, 2015-09-07:
Well thanks to the HistoryPlugin of imrahil it still gets shown as a "failure".
But don't worry I try to stay backwards compatible.

ntoff, 2015-09-07:
>If you cancel a print it obviously didn't run from start to finish

But it wasn't a failed print, it was a cancelled print. There's a difference. Failure means something went wrong, simply not wanting to wait for a calibration print to finish all the way when I have all the info I need isn't a failure.

> don't you think it a bit of a skewed cost/usefulness ratio to introduce modifier keys for canceling prints just to not have red file names in your file list? ;)

And don't take THIS the wrong way, but having "failures" listed when they aren't failures is also skewing the results.

Looks like I'm forking it and doing it myself then. Ho hum.

closed 2015-09-07 by ntoff

foosel, 2015-09-07:
OctoPrint itself isn't listing them as failures in the UI anywhere. It's only marking them red.

ntoff, 2015-09-07:
orly?

http://i.imgur.com/g7nyAG3.png

foosel, 2015-09-07:
That is a 3rd party plugin that you installed.

ntoff, 2015-09-07:
Sorry to waste your time then, now I feel stupid

foosel, 2015-09-07:
Btw, I agree that the wording is off here and I better should have called it "cancelled" nearly three years ago when I added that. The thing is that that is not easily solved now without thinking about migration paths, API adjustments etc. So a lot of stuff that causes a lot of work in order to get right and not cause people's stuff to break.

What I'm arguing here is that from my experience in software development it seldomly pays off to tackle something like that, making the complexity grow and hence the overall maintenance overhead larger, if all that it achieves is cosmetics (if we are not only talking about cosmetics, correct me please, but I so far understood both you and @Salandora that you mostly had an issue with the stuff being red when in fact you just cancelled it very early because you forgot something/wanted to calibrate something), and in such cases it might make more sense to think about alternative solutions to the problem. E.g.:

  * A plugin that un-reds the stuff that is now red in the file list (visual change only) or
  * a plugin that adds a popup after a cancelled print asking you if that was a failed print or you merely thought better and if the latter removes the history entry

ntoff, 2015-09-07:
Doesn't matter. I could have sworn I wiped octoprint when I broke it, I thought it was running vanilla. I guess I made a backup or something. I've flashed, backed up, re-flashed and messed with it so much I've lost track of what I've done to it and which rpi is running what.

Ignore my insanity.

foosel, 2015-09-07:
If it's a comforting thought, that happens to me all the time thanks to constantly reflashing stuff. I've now started to mark up the micro sd cards with metallic opaque sharpies (the only stuff I could find that sticks to those tiny black beasts) because I simply lost track all the time what version of OctoPi/OctoPrint in which configuration etc was on which card.