OctoPrint/OctoPrint #1050: [Request] Can't disable files tab, gcode viewer and Terminal for guests.

geekgarage, 2015-09-07:
I'm running octopi as a free service for my customers so they can follow their product being printed and they are also getting a time-lapsed video with their product. But i need to be able to disable files tab so they can't download the files, gcode viewer so they can't recreate my desigh layer by layer and Terminal so they can't copy the gcode straight out from the terminal (all thsi is for guests), so that competitors cannot copy my work I've spent countless hours designing. I can't find a plugin that does this and the settings doesn't seem to support any settings regarding this.

foosel, 2015-09-07:
I'm not sure I understand your use case fully but - if you don't want your users to be able to see anything but basically the webcam feed, why don't you just give your users only the URL to the webcam feed and tighten down access to the proper OctoPrint instance? If you strip what you mentioned from the anonymous view of OctoPrint, there's not much left over anyhow, so unless I misunderstand something completely here, that doesn't make much sense.

geekgarage, 2015-09-07:
I know there isn't much left, but what there is left is the parts that would be okay for customers to see , like what is shown in the state window/tab and the webcam feed and timelaps videos. I really can't see why you would give guests access to the terminal view, gcode viewer and the file download in the first place. They are guests not users or administrators that potentially would have a need for that information and access to download files :)

foosel, 2015-09-07:
> I really can't see why you would give guests access to the terminal view, gcode viewer and the file download in the first place.

Because OctoPrint was created with sharing access to a 3D printer in mind, not with locking everything down to be used as an anonymously accessible 3D printing kiosk. People use it to manage/control their own printer or the printer within their hackerspace, in a controlled environment. I'd be surprised if most of the OctoPrint instances out there didn't actually only have one user configured and no port forward. To put it differently, I cannot see why someone would allow anyone access to their OctoPrint installation anonymously over the internet. I know that people do this and I try my best to keep the administrative and print control functionality locked down for that reason, but I'd not put my paper printer's web interface online for everyone to monitor either - you never know what people will try to do, and even if it just ends in DDOSing my printer I'd already have problems on hand. 

Your use case is simply a completely different one than what is the core usecase of OctoPrint. You want to use it (correct me if I'm wrong) as a view into your manufacturing business for people to see what they'll get from you, but without any other information. That's not what OctoPrint was created for, so it's no surprise you find it lacking in that department. If all that you want is basically a glorified webcam feed, it doesn't make much sense from both a security and a complexity point of view to cut down basically everything of an existing web interface (that offers full control over the printer) until you arrive at what you want, but instead it would make more sense to offer to your customers a minimal version that only provides the webcam feed and maybe a bit of progress information, thereby also limiting exposure to the actual control interface. That is the exact reason why there is a REST API and why the next version will also allow to offer different UIs on `/` through plugins.

foosel, 2015-09-07:
That being said, more control options with regards to user roles and denying more functionality for anonymous users is already tracked in #640

geekgarage, 2015-09-07:
Alright i see your point, haven't thought of it that way :) But then i would really like to know how i can let the webcam feed be accessible from the web but not the rest? I can't find any guides and it's not running on a specific port from what i can see, and I have never used haproxy, and i'm not that great at linux.

Just remember that not every one is a coder and know how to use the api and build a website with the info that they need. I would guess that most of the users of octopi is non-coders that know a little bit of linux (enough to get it running) and sees octopi as a webserver and not as a printer interface or a bit of both

I'm a small business, or more of a hobby company beside my education, I don't have hundreds or thousands of dollars to throw at a developer to make me a glorified webcam feed, i was looking at a one step solution, where there is some user control.

Ang getting from
http://octopi.local/webcam/?action=stream

to

http://sub.mydomain.dk/

The sub domain already have the record to point to my ip address at home where the printer is, i just don't have a clue where and what to setup on the pi to get the webcam feed, and at the same time, completely remove access from WAN to the rest of the interface. but when i call http://octopi.local i get the full interface.

foosel, 2015-09-07:
> Just remember that not every one is a coder and know how to use the api and build a website with the info that they need.

Yep, I am aware of that, which is why there is also a Mailinglist and a G+ Community to ask for help on, especially when it comes to things like: 

> i just don't have a clue where and what to setup on the pi to get the webcam feed

and why I'm also investing a LOT of time and effort to create a plugin system and an API so that people like you can get help from people who aren't only me for specialized requests. Please don't assume I'm in the "I don't care about people who can't code" camp just because I'm pointing out when specific problems are better solved separately - I also need to protect the scope of the software here, and trying to put in settings for every possible flavor of use case small groups of users or even just single users might be having is a surefire way to have a project die of feature bloat. Fun fact: Most people want the webcam hidden to guests but make available everything else, or want the gcode viewer active but the file download disabled (which would be nothing but snake oil) or other such permutations - accomodating everyone by single switches would end in insanity, so another approach at access control would be necessary instead, and for that there's the ticket linked above.

Finally, especially if you are running a small business, you should invest the time it needs to read up a bit on reverse proxies and especially firewalling though if you intend to expose your stuff on the net - you are basically running around with a loaded shotgun pointed at your feet otherwise. 

Now back to your problem: I'd be surprised if people wouldn't chime in to help you (or even quickly throw together a small web interface for you to use), without having to spend "hundreds or thousands of dollars" (a friendly advice, if someone charges you that much for something that's a static webpage with an embedded webcam feed, they are ripping you off).

I have no idea what router you have running at home, so it's a bit tricky to tell you what exactly to do, but what I'd suggest you to do is (assuming you have a port forward in your router setup that redirects incoming tcp port 8080 to octopi.local tcp port 8080) instead of redirecting your subdomain there completely to your router's external IP have it instead host a small http page that embeds your webcam feed (very basic would just be this: `<html><body>[image]</body></html>` in a file `index.html` somewhere on your webspace) and either remove the port forward for the full webinterface (octopi.local, tcp port 80) again or [configure the haproxy on octopi to at least require some additional credentials](https://github.com/foosel/OctoPrint/wiki/Setup-on-a-Raspberry-Pi-running-Raspbian#user-authentication). [*]

I also hope you don't mind me closing this ticket now since as already pointed out above, more granular options with regards to access control are already covered by another ticket. You can still comment though.

[*] Let me add that this is a very very simple solution to your problem with gaping holes. It would actually be way better to install some sort of caching proxy and possibly also a webcam multiplexer on your router or another system in the DMZ, or at least rate-limiting the access to the forwarded ports through corresponding firewalling rules in order to make sure that users with malicious intent connecting from outside cannot DDOS your Raspberry Pi - by directly port forwarding to it, it has to handle any request that gets sent to it, that may include malicious requests, effectively allowing anonymous people to get your printer come to a screeching halt if just enough requests need to be processed, more than the little CPU can handle effectively while also driving a print job.

closed 2015-09-07 by foosel