OctoPrint/OctoPrint #1052: Login is not working properly using Google Chrome

david-pacheco, 2015-09-08:
I'm testing the octoprint user access control, and after successfully login in the admin user using the Chrome browser, if I refresh the page right after the login, the session/login is lost. If I do the exact same procedure using Firefox it works...

foosel, 2015-09-08:
Please update, your development version is ancient (we are currently on 1.3.0-dev-391) . If the problem still reproduces under the current `devel` branch report back with full log & Javascript console (I saw that the logs don't seem to contain anything relevant, I'd like to see them anyhow).

david-pacheco, 2015-09-08:
Ok, apparently the issue was solved even in the version I mentioned. I was using the master version from July, and updated to the current master version. But probably due to some caching (maybe) the problem seem to have disappeared after restarting the application and refreshing the page a few times. All good for now.

closed 2015-09-08 by david-pacheco