OctoPrint/OctoPrint #1053: Request to Raise Print Head when Cancelling or Pausing Print Job

dcwalmsley, 2015-09-08:
I have been running into a issue when I pause a print and the print head stops.  The print area touched by the nozzle heats up to a point where it melts and becomes deformed similar to a crater affect.  
My Overlord Print has the ability to raise the head when paused locally and to pick up where it left off when I ask it to resume.
1.  I would like to see both Pause and Cancel functions immediately lift the extruder head up and away from printed material.
2.  Allow the printer to pickup where left off when deselecting Pause.

foosel, 2015-09-08:
1. can be done with [GCODE scripts](http://docs.octoprint.org/en/master/features/gcode_scripts.html), 2. is already tracked in #366 - or am I missing something here?

dcwalmsley, 2015-09-12:
My apologies with understanding GCODE scripts.  I have never worked with this process and was looking for a means to raise printer head then cancel or pause was initiated and when resuming have the printer pick up where it left off.  At this point, I do not know how to make this work.

markwal, 2015-09-12:
@dcwalmsley Go ahead and post this question about how to do this to the google group or in the OctoPrint community on google plus.  I'm sure somebody has theirs set up to do this.

foosel, 2015-09-12:
@dcwalmsley what I was trying to tell you is that what you want to do is only partially possible for now. You can add some gcode (see the documentation I linked) to have it raise the head on pause and lower it again on resume. The tricky part is setting up gcode to move the head back to the exact position from before the pause by OctoPrint, since due to technical reasons at the moment it simply doesn't know that position (see also the FAQ for this). Changing that however already has a ticket in #366, which makes this one a duplicate, which is why I'm going to close it now. 

Moving the head out of the way on pause will usually move it back to nearly the correct position however ones the next line from the printed file is sent to the printer upon resume, so you might want to experiment with that.

To summarize: Take a look into Settings > GCODE Scripts for one part of what you are requesting (if you have absolutely no idea what GCODE is, take a look [here](http://reprap.org/wiki/Gcode) and/or follow Mark's advice), but be aware that what you want cannot be done fully correctly at the moment but will be tracked in an existing ticket (which you may follow to stay updated).

closed 2015-09-12 by foosel