OctoPrint/OctoPrint #1054: Mac OSX Yosemite 10.10.5

jrpmedia, 2015-09-08:
I am not a power user for Unix, but have followed the install process to the letter here:
https://github.com/foosel/OctoPrint/wiki/Setup-on-Mac

I get the error mentioned:

Traceback (most recent call last):
  File "/Applications/OctoPrint.app/Contents/Resources/script", line 21, in <module>
    import octoprint
  File "/Applications/OctoPrint.app/Contents/Resources/src/octoprint/__init__.py", line 4, in <module>
    from octoprint.server import Server
  File "/Applications/OctoPrint.app/Contents/Resources/src/octoprint/server/__init__.py", line 5, in <module>
    from sockjs.tornado import SockJSRouter
ImportError: No module named sockjs.tornado

I follow the suggested solution:

sudo pip install -r requirements.txt

And I get:

$ sudo pip install -r requirements.txt
Password:
The directory '/Users/admin/Library/Caches/pip/http' or its parent directory is not owned by the current user and the cache has been disabled. Please check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
The directory '/Users/admin/Library/Caches/pip' or its parent directory is not owned by the current user and caching wheels has been disabled. check the permissions and owner of that directory. If executing pip with sudo, you may want sudo's -H flag.
Could not open requirements file: [Errno 2] No such file or directory: 'requirements.txt'

Please can you redirect, or perhaps update the instructions in this instance, for low-level Users like me?

Thank you.

jrpmedia, 2015-09-08:
I have moved to google groups, apologises.

foosel, 2015-09-11:
That's more a case of a support request, better suited to the Mailinglist or the G+ Community (as pointed out over on G+ as well, and as pointed out in the Contribution Guidelines linked to by the bot above). Closing this.

PS: I read through your ticket in the wanhao group, as already pointed out there, you probably rather want to install OctoPrint on a Pi or something anyhow. While you can install it on a desktop machine and use it that way, that will block your desktop machine. The key idea behind OctoPrint is that you install it on something small, cheap and embedded to control your printer then through any web browser, not having it run on your desktop PC and connect your printer to that, that doesn't really gain you anything.

closed 2015-09-11 by foosel