OctoPrint/OctoPrint #1055: [request] Please show Z-Height (again)

Lord2Vader, 2015-09-09:
In previus versions i had an indicator for the z height so i knew when to watch out for critical moments in the print such as bridges or thin walls.

Could you please refit it in the status section?

BR Vader

markwal, 2015-09-12:
I think this is a duplicate of #355

foosel, 2015-09-22:
And of #1046. There's a reason it got removed, outlined in #1046. The z-height that used to be reported in the UI was inaccurate most of the time.

If you want to be notified about certain "critical moments" in the print, something like injected custom codes inside those printed files combined with a gcode hook in a plugin might make more sense. 

Alternatively try [this plugin](https://github.com/foosel/OctoPrint-DisplayZ).

closed 2015-09-22 by foosel