OctoPrint/OctoPrint #1056: Rebuild Mac OctoPrint.app - current build missing API calls (and dependencies)

vallon, 2015-09-09:
I had some difficulty getting OctoPrint.app installed (on a Mac).  I was able to get pip to install all the dependencies into the App (--target), but then realized the the bundled OctoPrint was missing most of the API calls (Slic3r could not spool anything).  So, I had to rebuild the entire thing from scratch.

That turned out to be quite easy (though, not at first). PYTHONUSERBASE can specify an alternate "user" tree.  I pip installed the OctoPrint.src.zip into $app/Contents/Resources, and it installed a complete distribution (pkg + dependencies).

Here is the recipe:

1. Download OctoPrint.app.zip, unzip it.
2. Remove everything from Contents/Resources except AppSettings.plist, MainMenu.nib, appIcon.icns
3. export PYTHONUSERBASE=$appdir/Contents/Resources
5. pip install --user https://github.com/foosel/OctoPrint/archive/1.2.6.zip
6. New Contents/Resources/script:

```
#!/usr/bin/env python

import os
import sys
base = os.path.dirname(os.path.realpath(__file__))

os.environ['PYTHONUSERBASE'] = base
# pip --user for plugin pip
os.environ['PIP_USER'] = '1'

script = os.path.join(base, "bin/octoprint")
os.execl(script, script)
```

This gives an OctoPrint.app with all its dependencies installed inside the App bundle.  This would mean the App would run on all stock OSX installs (given python 2.7, etc).  Plugins management would need PIP (sudo easy_install pip), but that is straightforward.

I also saw somebody mention something about "virtualenv", but did not explore it, as I saw the reference after I had the 'pip --user' build working.

Is there any interest in this?  Is there an OctoPrint.app build script in git that I can patch?

foosel, 2015-09-10:
Hi @vallon 

First of all, yes, there's definitely an interest in this (at least by me). Now the tricky part... The app linked on the setup page was done by a user (it is an open wiki anyone with a Github account can edit, that does have pros and cons as you can see ;)), there are no sources for that in the official repo (which as far as possible should stay platform agnostic anyhow). I have absolutely no experience with Macs, I do have a Mac Mini here for testing if push comes to shove, but that's about it. So I cannot do anything about that stuff that I'd feel comfortable sharing and supporting.

If however you have something that works for you, I'd suggest we create a repo for that (if you want I can even give you one under the OctoPrint organization?) and make it transparent how that app file is built. 

If that sounds like complete nonsense, it's because I have no idea how apps work on Macs, I'm a Linux/Windows person. So I really need some input here how best to proceed - and volunteers to do the actual setup for the package, since well... you wouldn't want to depend on anything I've packaged up for a Mac due to the reasons listed above :)

vallon, 2015-09-10:
The wrapper-app is open source http://www.sveinbjorn.org/platypus, built with V4.7.  It takes a script (any exec-able thing) and some app support files and wraps it up into a launchable App that shows stdout/stderr in a terminal window (there are other modes as well).

Platypus can be given a config script, and it can be run at the command-line.  I think I can reduce the mac build to a build_mac.sh + OctoPrint.platypus (config file).  I'll keep simplifying it.

foosel, 2017-03-08:
@vallon are you still looking into this/did something come out of this? Doing a little spring cleaning of the bug tracker and wondering if there's a reason this is still open.

closed 2022-01-18 by foosel