OctoPrint/OctoPrint #1057: Plugin Update fails

jminardi, 2015-09-11:
1. What were you doing?
Attempting to update a plugin

2. What did you expect to happen?
List of all plugins to display

3. What happened instead?
The list was empty

4. Branch & Commit or Version of OctoPrint:
1.2.6

5. Printer model & used firmware incl. version
   (if applicable - always include if unsure):
N/A

6. Browser and Version of Browser, Operating 
   System running Browser (if applicable - always 
   include if unsure):
Latest Chrome OS X

7. Link to octoprint.log on gist.github.com or pastebin.com
   (ALWAYS INCLUDE AND DO NOT TRUNCATE):
http://pastebin.com/h50dT9wA

8. Link to contents of terminal tab or serial.log on 
   gist.github.com or pastebin.com (if applicable - always 
   include if unsure or reporting communication issues AND
   DO NOT TRUNCATE):
N/A

9. Link to contents of Javascript console in the browser 
   on gist.github.com or pastebin.com or alternatively a
   screenshot (if applicable - always include if unsure
   or reporting UI issues):

Synchronous XMLHttpRequest on the main thread is deprecated because of its detrimental effects to the end user's experience. For more help, check http://xhr.spec.whatwg.org/.
packed_app.js:14008 Starting dependency resolution...
packed_app.js:14065 ... dependency resolution done
packed_app.js:14284 Going to bind 25 view models...
packed_app.js:14352 ... binding done
http://10.24.1.76/plugin/softwareupdate/check Failed to load resource: the server responded with a status of 500 (INTERNAL SERVER ERROR)
http://10.24.1.76/webcam/?action=stream&1441933517107 Failed to load resource: the server responded with a status of 503 (Service Unavailable)

10. Screenshot(s) showing the problem (if applicable - always
    include if unsure or reporting UI issues):

[image]

I have read the FAQ.

Relevant Error:

```
2015-09-11 01:01:32,582 - octoprint - ERROR - Exception on /plugin/softwareupdate/check [GET]
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1817, in wsgi_app
    response = self.full_dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1477, in full_dispatch_request
    rv = self.handle_user_exception(e)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1381, in handle_user_exception
    reraise(exc_type, exc_value, tb)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1475, in full_dispatch_request
    rv = self.dispatch_request()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask-0.10.1-py2.7.egg/flask/app.py", line 1461, in dispatch_request
    return self.view_functions[rule.endpoint](**req.view_args)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/OctoPrint-1.2.6-py2.7.egg/octoprint/server/util/flask.py", line 437, in decorated_view
    return flask.ext.login.login_required(func)(*args, **kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/Flask_Login-0.2.2-py2.7.egg/flask_login.py", line 637, in decorated_view
    return func(*args, **kwargs)
  File "/home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.6-py2.7.egg/octoprint/plugins/softwareupdate/__init__.py", line 332, in check_for_update
    information, update_available, update_possible = self.get_current_versions(check_targets=check_targets, force=force)
  File "/home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.6-py2.7.egg/octoprint/plugins/softwareupdate/__init__.py", line 403, in get_current_versions
    populated_check = self._populated_check(target, check)
  File "/home/pi/oprint/lib/python2.7/site-packages/OctoPrint-1.2.6-py2.7.egg/octoprint/plugins/softwareupdate/__init__.py", line 674, in _populated_check
    if check["type"] in ("github_commit"):
KeyError: 'type'
```

foosel, 2015-09-11:
Some plugin that you installed is returning an incomplete update definition here. OctoPrint doesn't capture that correctly (that is a problem to solve in OctoPrint to prevent stuff like this to be caused by incorrectly implemented plugins), however you should also figure out which plugin that is uninstall it and report it to the plugin author (or fix it yourself if that author happens to be you).

foosel, 2015-09-11:
Fixed in `devel` and `maintenance`, going to be released with 1.2.7.

commit 2efc5c4f (2015-09-11): Ignore update definitions that are lacking the type

Caused a KeyError so far, update definitions that are broken like that
will now just be ignored instead.

Closes #1057
commit b56ba658 (2015-09-11): Ignore update definitions that are lacking the type

Caused a KeyError so far, update definitions that are broken like that
will now just be ignored instead.

Closes #1057
(cherry picked from commit 2efc5c4)
closed 2015-09-11 by foosel