OctoPrint/OctoPrint #1061: Missing requirements-bbb.txt?

DirtC, 2015-09-12:
Followed these instructions precisely:
https://github.com/foosel/OctoPrint/wiki/Setup-on-BeagleBone-Black-running-%C3%85ngstr%C3%B6m

AT THIS POINT:
git clone http://github.com/foosel/OctoPrint
cd OctoPrint
beaglebone:~/OctoPrint$ ls
AUTHORS.md    CONTRIBUTING.md  MANIFEST.in  babel.cfg  requirements.txt  scripts   src    translations
CHANGELOG.md  LICENSE          README.md    docs       run               setup.py  tests  versioneer.py

requirements-bbb.txt is missing. I am assuming it is meant to be in this folder with the git pull?

I tried creating one using this:
http://code.openhub.net/file?fid=ENPFtnyZ5dUj7getPvKWq8Ard9M&cid=p7TxIddQRZk&s=&fp=413159&mp=&projSelected=true#L0

But that generates a build error.

foosel, 2015-09-13:
Apparently I forgot to adjust the guide... Try the regular requirements.txt
file please.

DirtC, 2015-09-13:
Turns out there was a little more to it.LOVE IT, so thanks so much for the work!Here's what I did to get it working:

   
   - Get a power supply and make sure your board always has power.
   - Connect it to the Internet. Skipping this step will lead to heartache. Don’t think you can skip it.
   - Get the latest version of Angstrom here:https://s3.amazonaws.com/angstrom/demo/beaglebone/BBB-eMMC-flasher-2013.09.04.img.xz      
      - Follow these instructions
      - Note: you can probably get this running on Debian – I didn’t try.
      - Note: DNS stopped working at weird times with WGET throwing errors If so, reboot.

   - At this point your board has rebooted and you’re connected via Putty. The next steps are going to take a while, so get some tea or coffee.
   - Logged in as root, for now.   
Make sure your system time is set correctly. This is a big source of failure.   
date   
Sun Sep 13 21:42:11 UTC 2015   
If the date is not correct, make sure to set up NTP   
http://codepen.io/timbergus/timbergus/bbbangnode
Update Packages
opkg update ; opkg --force-reinstall install connman-angstrom-settingsopkg update
Takes a while, then update the tools. If you don’t you will get version conflict errors.
pip install --upgrade setuptools
Update CA Certs
opkg install ca-certificates
Add octo as a user and give it some perms it needs:
useradd octo
passwd octo [enter password]
usermod -a -G tty octo //add octo to tty
usermod -a -G dialout octo //add octo to dialout
usermod -a -G sudo octo //add octo to sudo 
Follow this precisely. Make sure everything finishes and there are no errors.
cd ~
su
opkg install python python-modules python-pyserial python-numpy python-setuptools python-misc python-pip git
exit
git config --global http.sslverify false (NOTE: this works after updating SSL, if you did that step – do it if you did not.)
git clone http://github.com/foosel/OctoPrint
cd OctoPrint(case matters!)
su
pip install -r requirements.txt (note this is NOT requirements-bbb.txt -it does not exist, although everything in it seems to be now in another file)
python setup.py install
exit (you should be running it as octo after this point)
./run

koenkooi, 2015-09-14:
FWIW, with angstrom v2014.12 and newer you can just do 'opkg update ; opkg install octoprint-nginx' to get a working octoprint install. Alternatively there are rootfs tarballs in the nightlies section: http://dominion.thruhere.net/angstrom/nightlies/v2014.12/beaglebone/ 

No flasher images yet, sorry.

DirtC, 2015-09-14:
I blame Google :) 

Ok, thanks a million for this. At least I spent time getting my brain around the device - my first BBB.

Maybe it's time for those other instructions to be removed or updated? I don't know if I'm supposed to do an edit. Will happily do so.

koenkooi, 2015-09-14:
I haven't found the time/motivation to update the wiki, so please go ahead :)

foosel, 2015-09-14:
I fear I was too fast, did it a couple minutes ago ;) but better double check :)

This can be closed I assume?

DirtC, 2015-09-14:
Let me try koenkooi's solution tonight and then I'll update the wiki, then we close? Your call. Don't have time right now.

foosel, 2015-09-14:
Fine by be as well

DirtC, 2015-09-16:
@foosel Ok, so I had to update my proccess a bit, but I think it now works (I think I tried a variety of things and wrote them down in the wrong order originally). Anyway, I will test one more time to make sure.

@koenkooi How do I get a build onto my BBB without the flasher? Google failed me again.

foosel, 2015-11-15:
What's the status here?

foosel, 2015-11-20:
Considering this closed.

closed 2015-11-20 by foosel