OctoPrint/OctoPrint #1062: Error installing plugins using Archlinux (pip2)

mcanes, 2015-09-13:
1. What were you doing?
Installing a plugin using the web interface plugin manager. Octoprint prompt a generic error.
```Something went wrong
Please consult octoprint.log for details```

2. What did you expect to happen?
Install the plugin. The plugin log printed this message.
```Installing plugin "EEPROM Marlin Editor" from https://github.com/amsbr/OctoPrint-EEprom-Marlin/archive/master.zip...
Done!```

3. What happened instead?
Octoprint try to find the command *pip* but in archlinux pip for python2 is *pip2*.
So OctoPrint can't find the pip executable and failed.

4. Branch & Commit or Version of OctoPrint:
Version: 1.2.6 (master branch)

5. Printer model & used firmware incl. version
   BCN3d+ with 0x00 firmware (marlin with some changes)

6. Browser and Version of Browser, Operating 
   System running Browser (if applicable - always 
   include if unsure):
Chrome  45.0.2454.85 (64-bit)
Running on Linux.

Octoprint running on RPI with archlinux.

7. Link to octoprint.log on gist.github.com or pastebin.com
   (ALWAYS INCLUDE AND DO NOT TRUNCATE):
http://pastebin.com/DX83S3Ly

8. Link to contents of terminal tab or serial.log on 
   gist.github.com or pastebin.com (if applicable - always 
   include if unsure or reporting communication issues AND
   DO NOT TRUNCATE):

9. Link to contents of Javascript console in the browser 
   on gist.github.com or pastebin.com or alternatively a
   screenshot (if applicable - always include if unsure
   or reporting UI issues):

10. Screenshot(s) showing the problem (if applicable - always include if unsure or reporting UI issues):

Notes:
Maybe we can use *import pip* and use the pip module instead of the pip command line.

Thanks.

foosel, 2015-09-13:
You can override the pip command used to install plugins, [here](https://github.com/foosel/OctoPrint/wiki/Plugin:-Plugin-Manager#configuring-the-plugin).

At the very beginning I in fact did use pip as a library like you suggest. I can't remember the exact problems I ran into with that that finally broke the camel's back, but I do remember that there were a lot of issues that were solved by switching to the external call approach that is now used.

foosel, 2015-09-13:
Marked as request to add configuration dialog for pip command

foosel, 2015-09-22:
Config dialog added in both `devel` and `maintenance` branch, will be included in 1.2.7 release.

closed 2015-09-22 by foosel