OctoPrint/OctoPrint #1063: Password reset

kareem613, 2015-09-16:
It would be good to have the ability to reset a forgotten password. 
Even a wiki entry on how to do it from the command line would cut it. 

Helps avoid being completely locked out of the interface.. like I am right now. :)

markwal, 2015-09-16:
https://github.com/foosel/OctoPrint/wiki/Configuring-Access-Control

Near the bottom, it says to rerun the wizard, do the following:
1. edit ~/.octoprint/config.yaml and set `firstRun: true`.  It should be there already set to `false` in your case.
2. remove ~/.octoprint/users.yaml

I would also: `sudo service octoprint restart`

kareem613, 2015-09-16:
Perfect! Thanks. I guess this isn't a feature request then.

foosel, 2015-09-22:
@kareem613 as the bot kindly told you, it wasn't marked as a feature request by you in any case ;)

closed 2015-09-22 by foosel

flowchartsman, 2018-01-31:
If anyone is curious, there's a shortcut way to do it, so long as you have UI access.  Simply go to settings -> server and change the "Restart System" command from `sudo shutdown -r now` to `echo "pi:pi" | sudo chpasswd`, then click <power button> -> "Reboot System" -> Proceed.  Just make sure to change the restart command back when you're done.  No telling how long this will work, as it's definitely a security issue and should probably be fixed.

foosel, 2018-01-31:
That being possible is caused by passwordless sudo being allowed in Raspbian and hence OctoPi incl 0.14 by default, and that is addressed in guysoft/OctoPi#345

foosel, 2018-01-31:
And after having read up on the ticket again, this is also not related to the login credentials discussed here ;)

Glastis, 2020-05-23:
Here is a one-liner pass reset: 
```
sed -i "s/password: $(echo 'NEW_PASSWORD' | sha512sum | awk -F ' ' '{print $1}')\npermissions/password:*\npermissions/" ~/.octoprint/users.yaml
```
Just change NEW_PASSWORD by what you want