OctoPrint/OctoPrint #1064: Controlling a WeMo Switch rather than relay board from your RPi   [Request]

julian800, 2015-09-16:
Hi,
I am planning on controlling power on/off my 3D printer from RPi. I did find a page about Controlling a relay board from your RPi by TwystNeko. I am not really confident to modify power cable of 110V or 240V. I found a possible solution for myself  "Belkin WeMo switch". It is Wi-Fi enabled Switch lets you turn electronic devices on or off from anywhere. I know there is a python Open source control for Belkin WeMo devices: ouimeaux 0.7.9   Could we have a feature for (- action: printer on  and - action: printer off )  sending control command to WeMo switch or document to explain setup of this? 

Thanks

jminardi, 2015-09-18:
You can use a system command to call your python script: https://github.com/foosel/OctoPrint/wiki/Configuration#system

Grogyan, 2015-09-18:
I am planning to do something similar.  However, I opted to use an ATX
power supply and have ha transistor control the on/off from the Arduino
instead of the Raspberry PI, as there is already a gcode to do this.  Then
it is just a simple matter of updating config.yaml as documented on the
OctoPrint wiki

foosel, 2015-09-22:
This is already possible with system commands, as outlined by @jminardi above (although the linked wiki page is for older versions of OctoPrint, documentation of 1.2.0 resides at [docs.octoprint.org](http://docs.octoprint.org/en/master/configuration/config_yaml.html#system)). No need to add anything extra within OctoPrint here.

closed 2015-09-22 by foosel