OctoPrint/OctoPrint #1069: exception in tornado module

TonyLHansen, 2015-09-21:
New installation of 1.2.6 on a Raspberry Pi using OctoPi distribution.

I'm getting these errors in the log:

2015-09-20 21:36:35,467 - tornado.application - ERROR - Uncaught exception
Traceback (most recent call last):
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado-4.0.1-py2.7-linux-armv7l.egg/tornado/http1connection.py", line 234, in _read_message
    delegate.finish()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado-4.0.1-py2.7-linux-armv7l.egg/tornado/httpserver.py", line 282, in finish
    self.delegate.finish()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado-4.0.1-py2.7-linux-armv7l.egg/tornado/web.py", line 1884, in finish
    self.execute()
  File "/home/pi/oprint/local/lib/python2.7/site-packages/tornado-4.0.1-py2.7-linux-armv7l.egg/tornado/web.py", line 1914, in execute
    self.handler._execute(transforms, *self.path_args, **self.path_kwargs)
  File "/home/pi/oprint/local/lib/python2.7/site-packages/sockjs_tornado-1.0.1-py2.7.egg/sockjs/tornado/websocket.py", line 44, in _execute
    self.stream.write(escape.utf8(
AttributeError: 'NoneType' object has no attribute 'write'

Line 44 of tornado/websocket.py has this code:

        # Upgrade header should be present and should be equal to WebSocket
        if self.request.headers.get("Upgrade", "").lower() != "websocket":
            self.stream.write(escape.utf8(
                "HTTP/1.1 400 Bad Request\r\n"
                "Connection: Close\r\n"
                "\r\n"
                "Can \"Upgrade\" only to \"WebSocket\"."
            ))
            self.stream.close()
            return

foosel, 2015-09-21:
Does your installation work otherwise? You haven't stated that. If so, those messages are caused by a client on your network closing the connection "in the middle of speaking" and can be safely ignored.

foosel, 2015-09-22:
Well, in any case I've updated the dependencies to a new version of Tornado and SockJS-Tornado (the stack trace you pasted above is within those two components, OctoPrint code doesn't even get touched) and from the looks of it those new versions are more error resilient for exactly this specific case, so that should be solved in `devel`/v1.3.0, I might also backport it to 1.2.x, not sure yet.

closed 2015-09-22 by foosel

TonyLHansen, 2015-09-22:
Sorry, I haven't been near my printer in order to answer your questions. The next time I file a bug report here, I'll try to follow the guidelines more thoroughly. The problem I had is that connecting to the printer just hangs, and the only error messages I found anywhere were these dealing with Tornado.

Thanks for improving the Tornado code processing. Hopefully this improvement will fix my problems.

foosel, 2015-09-22:
Probably not, your installation of OctoPrint not being able to connect to your printer sounds more like wrong baudrates, serial ports or general protocol issues, for which the `serial.log` and/or the contents of the Terminal tab is what you want (which is among the things asked for in the bug reporting template in the contribution guidelines).