OctoPrint/OctoPrint #1070: [BRAINSTORM] E-Mail Support?

dcwalmsley, 2015-09-24:
Is there a use for email support within OctoPrint?  My simple answer would be yes.
Email should be added to inform user when a occurrence has happened such as when a print job  has completed, been paused, cancelled, and or interrupted by any number or reasons such as power, network, or human interruptions.  Email should contain explanation if possible to any items listed above and settings within OctoPrint should allow user to select what alerts they wish to be notified of.

In my humble opinion, OctoPrint  supports internet monitoring as a means to keeping an eye on the print, but not everyone has time to keep an active screen up and an email notification process will certainly alert the user that an event has changed with the print job.

foosel, 2015-09-24:
Perfect timing really, I just merged the PR to list [this plugin](https://github.com/anoved/OctoPrint-EmailNotifier) on the [Plugin Repository](http://plugins.octoprint.org).

dcwalmsley, 2015-09-24:
Good start, but I didn't see any options for what type of message you will receive other than the canned message the user would enter in the Message Content section.  Are there plans to expand the options to cover some or all the points I called put above?

foosel, 2015-09-24:
Those are questions better asked from the plugin author, who is @anoved, not me :)

In any case, any notification functionality down the road will always only be added through plugins, that makes stuff way more maintainable and flexible in the long run. 

The idea is to not turn the core application into a omnipotent behemoth nobody can navigate anymore, but instead allow users to mix and match the functionality they need. Plus also make it easier for people not part of the core development team to easily add the functionality they want and need.

That being said, OctoPrint can only ever allow plugins to notify about things that it knows about itself. Things like arbitrary print interruptions by external factors (print lifted, cat jumped on the print bed, PSU is on fire) are not really detectable in a generic kind of way, and certainly not only in software (what OctoPrint is after all) and would necessitate control and monitoring hardware individually added by the user for their specific setup which could then of course be attached to OctoPrint through - you might have guessed it - individual plugins, and through the concept of [helpers](http://docs.octoprint.org/en/master/plugins/concepts.html#helpers) that might also allow mails to be sent through something like the above linked plugin on such events.

anoved, 2015-09-24:
Adding email notification support for other events is certainly [on my radar](https://github.com/anoved/OctoPrint-EmailNotifier/issues/1). Despite the limitations @foosel mentions there are many [known events](http://docs.octoprint.org/en/master/events/index.html) that could trigger notifications. Expanding the settings interface to select which events and customize the messages for each is the main challenge for me.

dcwalmsley, 2015-09-24:
Much appreciated.  Looking forward to seeing enhancements.

R,
Doug Walmsley

foosel, 2015-09-25:
Closing this for now, I suggest subscribing to the repository of the EmailNotifier plugin to keep up to date on its development and/or contributing.

closed 2015-09-25 by foosel