OctoPrint/OctoPrint #1071: GCode Viewer doesn't show anything

dragondgold, 2015-09-25:
1. What were you doing?

Running a print through the web interface trying to view something on the GCode Viewer

2. What did you expect to happen?

See my model in the GCode Viewer

3. What happened instead?

The GCode Viewer is blank, it doesn't show my model it only shows some small lines when enabling `Show moves`.

[image]

4. Branch & Commit or Version of OctoPrint

Version: 1.2.6 (master branch)

5. Printer model & used firmware included version:

M3D Micro with M3D-Fio plugin for Octoprint

6. Browser and Version of Browser, Operating System running Browser:

Chrome on OS X Yosemite 10.10.5

7. Link to octoprint.log:

http://pastebin.com/XFGd3D1k

8. Link to contents of Javascript console:

http://pastebin.com/Fyga2H2m

foosel, 2015-09-25:
That Javascript log implicates that the loaded model has a couple thousand layers with absolutely no extrusions in it. Could you provide a GCODE file with which reproduction of the issue is possible?

dragondgold, 2015-09-25:
@foosel Sorry for the inconvenience, it seems that the problem is caused by the M3D-Fio plugin I'm using in order for octoprint to support my printer. The model has only 27 layers, M3D-Fio does some processing when uploading a .gcode file and seems to corrupt it. Thank you and sorry!

closed 2015-09-25 by dragondgold