OctoPrint/OctoPrint #1075: Cannot run pip with sudo

dragondgold, 2015-09-28:
1. What were you doing?

Modifying `config.yaml` in order to change the default `pip` command to `sudo /usr/bin/pip2` as stated in the docs here: https://github.com/foosel/OctoPrint/wiki/Plugin:-Plugin-Manager#configuring-the-plugin

2. What did you expect to happen?

To be able to install new plugins using the plugin manager because I need `sudo` in order to do it.

3. What happened instead?

The plugin manager is not able to detect pip. If I use `pip: '/usr/bin/pip2'` then `pip` is detected, but when using `pip: 'sudo /usr/bin/pip2'` it's not. I see in the log (attached below) that there is an `OSError: [Errno 13] Permission denied` but I don't know what permission is referring to, I already edited the `sudoers` file to allow running `pip2` without password.

4. Branch & Commit or Version of OctoPrint:

Latest `devel` branch as 28 September of 2015

5. Printer model & used firmware incl. version
   (if applicable - always include if unsure):

M3D Micro with M3D-Fio plugin for Octoprint

6. Browser and Version of Browser, Operating 
   System running Browser (if applicable - always 
   include if unsure):

Chrome 46.0.2490.33 beta (64-bit) on OS X Yosemite 10.10.5

7. Link to octoprint.log on gist.github.com or pastebin.com
   (ALWAYS INCLUDE AND DO NOT TRUNCATE):

http://pastebin.com/zapTMVWk

8. Link to contents of terminal tab or serial.log on 
   gist.github.com or pastebin.com (if applicable - always 
   include if unsure or reporting communication issues AND
   DO NOT TRUNCATE):

Just lots of `Recv: wait`

9. Link to contents of Javascript console in the browser 
   on gist.github.com or pastebin.com or alternatively a
   screenshot (if applicable - always include if unsure
   or reporting UI issues):

http://pastebin.com/CqWX8bFe

10. Screenshot(s) showing the problem (if applicable - always
    include if unsure or reporting UI issues):

[image]

I have read the FAQ.

foosel, 2015-09-28:
> To be able to install new plugins using the plugin manager because I need sudo in order to do it.

I **strongly** suggest to switch to using a user-owned virtual environment so that this is NOT necessary. Since plugins can basically contain anything, running their install routines as root might not be the best idea.

That being said, I'll look into fixing the call so that it allows sudo prefixing again (and add a warning that this is a bad idea in the process...)

dragondgold, 2015-09-28:
@foosel Thank you! I will use a virtual environment as suggested!

foosel, 2015-09-28:
This has now been fixed in `devel` and `maintenance` and will be part of 1.2.7.

Big fat warning is in place:

[image]

foosel, 2015-09-28:
Let me also add a big thank you for the fully fleshed out bug report :)

closed 2015-09-28 by foosel

dragondgold, 2015-09-28:
@foosel thank you for such a great piece of software :)