OctoPrint/OctoPrint #1076: Pausing job from REST API is not working

dragondgold, 2015-09-28:
1. What were you doing?

Sending a pause command from the REST API as stated here http://docs.octoprint.org/en/devel/api/job.html. I get the 204 Response but the job is not paused.

2. What did you expect to happen?

Job to be paused

3. What happened instead?

The current job is not paused, but I can pause it manually in the GUI

4. Branch & Commit or Version of OctoPrint:

Latest `devel` branch as 28 September of 2015

5. Printer model & used firmware incl. version
   (if applicable - always include if unsure):

M3D Micro with M3D-Fio plugin for Octoprint

6. Browser and Version of Browser, Operating 
   System running Browser (if applicable - always 
   include if unsure):

Chrome 46.0.2490.33 beta (64-bit) on OS X Yosemite 10.10.5

7. Link to octoprint.log on gist.github.com or pastebin.com
   (ALWAYS INCLUDE AND DO NOT TRUNCATE):

Nothing wrong

8. Link to contents of terminal tab or serial.log on 
   gist.github.com or pastebin.com (if applicable - always 
   include if unsure or reporting communication issues AND
   DO NOT TRUNCATE):

Just lots of `Recv: wait`

9. Link to contents of Javascript console in the browser 
   on gist.github.com or pastebin.com or alternatively a
   screenshot (if applicable - always include if unsure
   or reporting UI issues):

Nothing wrong

10. Screenshot(s) showing the problem (if applicable - always
    include if unsure or reporting UI issues):

No applicable

I have read the FAQ.

Salandora, 2015-09-29:
I took a fast look at this. I checked the way the UI is doing a pause, it is using the same way you linked to. 

I assume you have checked that the printer hasn't paused by not looking at the UI but the fact that the printer just won't stop printing.

Well I also assume you are using your octocmd repo to send this right?

dragondgold, 2015-09-29:
@Salandora Yes, the printer simply don't stop. But, when I click Pause in the GUI and then send a pause command from the REST API the printer does one or two movements and then stops again, it doesn't even resume the printing. I'm using my own python script, something like:

`session.post('http://localhost:5000/api/job', json={'command': 'pause'})`

Salandora, 2015-09-29:
hmm that's interesting, sound like octoprint would get 2 requests, (1 which resumes the print and 1 which imediatly pauses it again)

Any chance to see more of your python script?

dragondgold, 2015-09-29:
@Salandora here is the complete class I'm using http://pastebin.com/fv2TqWsQ. Then I just run `pause_job()`

closed 2015-09-29 by Salandora